markmal/freeship-plus-in-lazarus #2: Installation Errors

tvld, 2015-05-30:
Tried to install in Ubuntu 14.04. But this output:
./install-user.sh 
Installation FreeShip into user home
  Install executables
cp: cannot stat ‘Manuals’: No such file or directory
cp: cannot stat ‘Ships’: No such file or directory
  Install configuration
  Install Menu
xdg-icon-resource: file 'freeship-16.png' does not exist
xdg-icon-resource: file 'freeship-24.png' does not exist
xdg-icon-resource: file 'freeship-32.png' does not exist
xdg-icon-resource: file 'freeship-48.png' does not exist
  Install MIME
xdg-icon-resource: file 'freeship-16.png' does not exist
xdg-icon-resource: file 'freeship-16.png' does not exist
xdg-icon-resource: file 'freeship-24.png' does not exist
xdg-icon-resource: file 'freeship-24.png' does not exist
xdg-icon-resource: file 'freeship-32.png' does not exist
xdg-icon-resource: file 'freeship-32.png' does not exist
xdg-icon-resource: file 'freeship-48.png' does not exist
xdg-icon-resource: file 'freeship-48.png' does not exist
  Update Desktop database
  Update MIME database
Done
FreeShip is installed into /home/tom/FreeShip
  to uninstall enter into /home/tom/FreeShip and execute uninstall-user.sh

tvld, 2015-05-30:
Ok. Installed and it is running: I copied the files in the root of the ZIp file to the Exec folder
And of course needed to run:

*sudo apt-get install libqt4pas5*

markmal, 2015-06-06:
Hi Tom,
Thank you for installation and reporting problems.
Sorry for late reply. I was on vacation.
I will take a look.
What is your desktop environment? Unity?

thanks,
Mark

tvld, 2015-06-08:
Yes, Unity. Would it be much of an hassle to make a ppa for this?  I suppose the dependencies and installation folders would not cause any problem anymore...

markmal, 2015-06-09:
Hi Tom,

I am not familiar with PPA, just took a quick look, it is purely Ubuntu 
thing(?) and some LaunchPad should build a package from sources, if I 
understand it correctly. Currently Ubuntu has quite outdated Lazarus 1.1 
or so, I use 1.4R2. I am not sure if LaunchPad will compile it.
Also target of the project is to let it work on any Linux.

I've worked the issue around by adding required files and dirs to the 
zip. You can download v 3.5.2.58. 
https://github.com/markmal/freeship-plus-in-lazarus/releases/tag/3.5.2.58

thanks,
Mark

tvld, 2015-06-09:
Hi Mark, PPA is a Debian thing and many installations are based on that Linux distribution, such as Ubuntu. It is designed for secure and easy installation and upgrading. In Ubuntu you can also move your PPA to the Software centre, for example: https://apps.ubuntu.com/cat/applications/oneiric/sailcut/     It lowers the threshold for non-programmers tremendously.

That said, I have never done any PPA myself as I've not programmed native Linux.

speters, 2015-06-09:
I made some provisions to build a Debian package in my "debian" branch https://github.com/speters/freeship-plus-in-lazarus/commit/0b52e7f1493bbbeec31d5303bd5b2edd86820fbf

It compiles using
```
fakeroot debian/rules clean
fakeroot debian/rules build
fakeroot debian/rules binary
```
The resulting deb package is available at https://github.com/speters/freeship-plus-in-lazarus/releases/tag/3.5.2.58-1

But it still requires some manual work to set up the changelog, and version numbers when building a new version. Maybe a "checkinstall" based approach is easier...

markmal, 2015-06-10:
Thanks Sönke! 
Sounds cool

tvld, 2015-06-10:
Excellent. I immediately tried it... and it installed beautifully. Just got " missing language" and there are no ship models to play with. Where can I submit bugs for the .deb?

closed 2015-06-26 by markmal