markmal/freeship-plus-in-lazarus #4: HiDpi screens show very small icons

tvld, 2015-06-10:
I have a Lenovoa Yoga Pro 2, 3200x1800, and the icons are oops... small...:

[image]

tvld, 2015-06-10:
[image]

markmal, 2015-06-11:
> small_icons

Yes, icons are small. We need to make them bigger.

tvld, 2015-06-11:
On lower-res screen they are ok. I suppose two options: or scale automatically with screen dpi, or have a setting somewhere to adjust icon-scaling. The last option might be most universal for various distributions?

markmal, 2015-06-11:
Simple upscaling will make icons look pixelated and ugly. I think, new set of bigger icons has to be created manually.

tvld, 2015-06-11:
yeah. I might volunteer for a set of .svg icons?  These you can scale to any size

markmal, 2015-06-12:
Excellent idea, Tom! Welcome aboard!
I am not sure yet if SVG icons are supported in Lazarus/Qt controls directly, anyway, if not, we can convert them to bitmaps anytime in run time, when program starts.

tvld, 2015-06-18:
First attempt: 
https://github.com/tvld/freeship-plus-in-lazarus/tree/master/Icons

markmal, 2015-06-19:
Hi Tom!
thanks! looks cool.
I will start to make toolbar resizeable

tvld, 2015-06-19:
ok, then will proceed in same way. Would you need certain names for the icon files? If so, do you have a list?

speters, 2015-06-19:
Hi Tom,
great to see your work. A few days before, I also tried to draw some icons (https://github.com/speters/freeship-plus-in-lazarus/blob/master/resources/icons.svg), but I'm not satisfied with my results.

But it seems to be very hard to get it consistent, meaningful, and good looking at the same time. Even the guys from FreeCAD failed badly IMO (http://www.freecadweb.org/wiki/index.php?title=Artwork).

Have a nice day,
Sönke

tvld, 2015-06-19:
The trick is mostly to use standard, minimal palette. Freecad fails with too many colors IMHO. I have created this palette:  https://github.com/tvld/freeship-plus-in-lazarus/blob/master/Icons/FreeSheep-Standard-Palette.png   If you think it is ok, we can proceed.

tvld, 2015-06-19:
Sonke, Actually good work and we can syncrhonize. I will take your design and make separate icon files in the structure I suggested. Feel free to submit icons if you feel inclined, but let's agree we dont do the same based on our initial enthusiasm:))   If we agree on the choices.

Using Inkscape is easy and takes like no more than an hour for all icons I suppose.

tvld, 2015-06-19:
With the excellent icons, dont be modest, of Sönke, updated: https://github.com/tvld/freeship-plus-in-lazarus/blob/master/Icons/preview/Icons_sprite_preview_white.png
About 60% of icons I suppose ?

markmal, 2015-06-19:
Hi guys, great work!

question. Is background supposed to be white, or it will be transparent? 
If the latter, please also try how they will look on grey buttons.

I've investigated with SVG support in Lazarus (aggpas, fpvectorial, 
bgrabitmap), unfortunately it looks like there is just little support, 
for basic SVG.
No gradients, no transformations etc...
It means we should create SVG icons and convert them to PNG of sizes 
16,24,32,48,64,128 with using some 3-rd party tool.

Yes, icons should have names, I'll check.

tvld, 2015-06-19:
Hi Mark, Did you check out the folders https://github.com/tvld/freeship-plus-in-lazarus/tree/master/Icons ?  There are png in various sizes!  Also in preview you notice on dark backgr. Not ok now... Should we have universal icons, without transparency.. or a dark theme version?

tvld, 2015-06-19:
Committed a light background behind the icons to allow on dark themes. In  the future I suppose we should create a dark and white themed icons set. For now, this is my contribution ;)

markmal, 2015-06-19:
Tom, I like previews with both backgrounds. 
My question was, should we paint background in icons, or should be make it transparent.
I think we should have capability to use whatever theme, just by configuring FreeShip to use certain theme and size. We just need a naming convention for directories and files. Like you already have.
Like  icons/<theme>/[svg|size]
example:
icons/light/svg/file-open.svg
icons/light/png-16/file-open.png
icons/light/png-24/file-open.png
...
icons/dark/svg/file-open.svg
icons/dark/png-16/file-open.png
icons/dark/png-24/file-open.png
etc

I am not sure what is meaning of numbers in your file names. Should we have the numbers there?

tvld, 2015-06-20:
Mark, Freeship should draw the icons transparent I think. It depends on the desktop theme I suppose what people would like . Anyway, we can leave this decision to theme designers.

I like your idea about the theme structure. What would you think of this way:

/themes/default/icons/svg/file-open.svg
/themes/default/icons/icons/light/png-16/file-open.png
/themes/default/icons/icons/light/png-24/file-open.png
...

So we could add extra themes later on:

/themes/midnight-blue/icons/svg/file-open.svg
/themes/midnight-blue/icons/icons/light/png-16/file-open.png
...
WE can use an override mechanism: If a new theme did not define a certain icon, it would use the default which should always be present.

Thoughts?

markmal, 2015-06-20:
Tom,

I like idea with default theme!

I only don't understand "/icons/icons/light/png-24/" piece. Why "icons" 
is 2 times, what is "light"?
also lets use just "24" instead of "png-24", I think png is redundant 
here. should it matter for us what format files will be inside, png, or jpg?

I attach zip with names.
Please rename your images accordingly, it is very important, because 
names of icons should be named exactly as names of actions in program.
Sorry I am giving you names quite late. Did not have time during the week.

markmal, 2015-06-20:
zip file with original icons was not attached. I'll upload it somehow and let you know.

markmal, 2015-06-20:
I've placed it into release. please download https://github.com/markmal/freeship-plus-in-lazarus/releases/download/3.5.2.58/orig-icons-16.zip

tvld, 2015-06-21:
Sorry, the path structure was my bad: a cut and paste mistake...  I agree with the folder structure naming convention. 
The file names I needed indeed. Will use these and adapt my branch to it. Stay tuned ;)

tvld, 2015-06-21:
Pushed new folder structure with names.
Indeed I am missing quite a few icons. But.. I did not get your originals, your orig-icons-16.zip contains empty icons?

tvld, 2015-06-22:
Just pushed new icons with removed transparency: I did not realize most icons had a useless opacity of 96%.

markmal, 2015-06-22:
Hi Tom,
what do you mean by missing and by empty?

may be issue is that the icons are not PNG, they are actually BMP. I my 
linux it (extention) does not matter, it shows them any way.
I will rename and replace zip. You can rename too.

I've modified FreeShip to load icons from files, and to configure icon's 
dir and size.
It can use dirs from themes, but idea of themes is not expressed in 
design so far.

We should discuss, what parameters will be included into a theme, icons, 
what else? may be graphic colors? what else?

tvld, 2015-06-22:
Renaming to .bmp was solution indeed. Other theme stuff: maybe we could store background for toolbars = window decoration in standardised color definition file, for example in Json format?   I now used a palette for inkscape. I suppose it all depends on what you could configure in the program itself?

markmal, 2015-06-22:
Background for toolbar, as well as other controls look and feel comes from
OS settings. I am not sure why we should change it for one program. And it
will require quite amount of work.
Colors that are specific to FreeShip (like line colors etc..) are already
stored in ~/.config/FreeShip/FreeShip.ini file. I am not sure why we should
change it to Json?
Windows decorations are hard to change in X environment, it is in
WindowManager domain which may be on user's computer, where X Server runs,
and program can run on a different server computer (X Client). It is very
hard to change, if even possible, whatever is in X Server from X Client.

For now I would add these settings to a theme.
Each theme will have separate .ini file (i.e.Themes/Default/theme.ini)
In this file will be section [Graphics] where FreeShip colors will be
defined, and also is defined a size of icons (16,24,32,48,64,96,128) for
tools and for menu (menu icons are not supported in current Qt, but may
will be supported later. Menu icons are configured in program).
Icons of each size will be stored in subfolders: Themes/Default/16,
Themes/Default/24 etc...
All icons will have standard names (that I've sent in zip).
From beginning there will be configured Default theme with tool icon size
24.
User can reconfigure something and save as a custom theme.
How about this?

tvld, 2015-06-22:
Mark, my comments were without any understanding of the internal structure. What you suggest sounds fine to me.

tvld, 2015-06-23:
I would suggest to also implement theme.ini  with override mechanism. If we would not, adding a new setting to theme.ini sometime in future, would break a theme. Overriding would take the non-defined settings from the   /Themes/default/theme.ini

markmal, 2015-06-23:
Yes, and even more. A theme will have ParentTheme parameter. So themes 
can be born one of another inheriting properties.
Default -> SomeTheme -> AnotherSubtheme ...
Sort of OOP.
Currently I am working on inheritance of icon files. Colors just copied 
from a parent then some may be customized.
Icons can be re-defined in a child by placing file into its icon dirs.

tvld, 2015-06-23:
hm, dont get carried away I suppose? OOP sounds great but seems like you are putting a Ferrari Motor in an Volkswagen? )) 
Anyway, look forward to see the changes. Curious what it looks like :)

speters, 2015-06-26:
In my opinion, an OOP theme sytem looks far too complicated at the current point. If doing something like that, it would be better to make the ToolBar/ToolButtons be stylable via the standard Qt way of using stylesheets, as support for this already exists partially in the Lazarus codes (e.g. you can load a Qt stylesheet by starting with a commandline parameter `./FreeShip -stylesheet style.txt`, but this only works with widgets that are drawn by Qt).

In https://github.com/speters/freeship-plus-in-lazarus/pull/1 I tried a simple solution to make FreeShip load its ToolButton images from files, and adjust the ToolBar size accordingly.

[image]

The screenshot shows that the ToolBar is a bit crowded. Buttons like open-file, new-document, exit are of not much use there).

markmal, 2015-06-26:
> In speters#1
> freeship_largetoolbuttons

Hi Sönke,

Though currently the app is compiled with Qt, I am trying to make it 
widgetset-agnostic as much as possible.
Qt is chosen only for its out-of-the-box MDI windows support.
I've already developed "OOP" (not really OOP, rather multi-generation) 
schemes support.
Settings are done in GUI and stored in .ini files.
It pretty much works, now I am testing... As soon as it will pass my 
simple tests I'll publish it in git soon.

Yesterday I've re-captured original icons with transparency, finally 
they look correct. :) And resized them to all bigger sizes 
24,32,48,64,96,128.
They do not look perfect, but are visible.

Currently I am awaiting from Tom some subset of icons to be created and 
named accordingly, that I could create "Tom Theme" as child of "Default".

Yes, toolbar is crowded. I have an idea to split it to multiple 
sub-toolbars by groups, and make them wrappable separately, and may be 
dockable, so they can be moved to side of screen and do not occupy 
vertical space. (I think horisontal space on our wide-screen displays is 
less valuable)
What do you think?

tvld, 2015-06-26:
Hi Mark... Why did you had to recreate all the icons, I pushed them 4 days ago? They should be all there in good quality?  https://github.com/tvld/freeship-plus-in-lazarus/tree/master/Themes/default/Icons

markmal, 2015-06-26:
I've recreated the original icons. Those I've captured from the program. (they are embedded into ImageList inside the program, not stored separately in files.) 
Yours are ready? I missed this :) How about keeping both sets? I'll call the theme "Original" :)

speters, 2015-06-26:
FYI: I used `convert resources/tile_* -transparent black -depth 8 -scale 32x32 ImgList%d.png` to convert the original buttons to 8bit RGB with alpha/transparency. Paletted PNG didn't work for me.

markmal, 2015-06-26:
I've placed orig-icons.zip into  last release https://github.com/markmal/freeship-plus-in-lazarus/releases/download/3.5.2.58/orig-icons.zip
I'll include it as "Original" theme. Tom's theme will be "Default".
I like the way you've done toolbars, will merge it into trunk.

tvld, 2015-06-26:
@ mark: Well.. took me a lot of hours to get them right. Better should not have started.
@speters: with that you miss a lot. Please refer to the the convert commands in the script I developed for this.

markmal, 2015-06-26:
FreeShip with themes and configurable icon size is released.
https://github.com/markmal/freeship-plus-in-lazarus/releases/tag/3.5.3.1

markmal, 2015-07-01:
Hi Tom, I'm working on toolbar changes. (changing the original single toolbar to multiple toolbars (i.e. Files, Visibility, etc..) that wrap their buttons separately when form is resized.

I've removed Precision control. And ddded Undo, Redo and DeselectAll buttons.
Could you create icons for these new buttons: Undo.png, Redo.png and DeselectAll.png?

markmal, 2015-07-02:
For Intersection DIalog also need icons "CloseDialog", "AddOne", "AddRange", "DeleteAll"
[image]

markmal, 2015-07-02:
For Expaned Plates Dialog need more icons (some exist)
[image]

markmal, 2015-07-02:
Icons for TFreeLinesplanFrame
[image]

tvld, 2015-07-02:
Good progress:) .. Will be happy to add the missing icons. Sadly, broke my foot and the issues around it taking a bit of energy.  Will be back though. Need a week or so.

markmal, 2015-07-02:
Sad to know :( 
Get well soon!

markmal, 2015-07-05:
Hi Tom, 
For calculation dialogs we need three more icons: Ok.png, Cancel.png and Calculate.png

tvld, 2015-07-05:
Hi Mark, Added all others ( I hope... ;) ) 
Any still missing?

markmal, 2015-07-06:
Thanks Tom! I'll check it.

markmal, 2015-09-22:
I think it can be closed.

closed 2015-09-22 by markmal