markmal/freeship-plus-in-lazarus #5: Shading model too slow

tvld, 2015-06-10:
Seems we are not using the correct hardware accelerated libraries?  Shading and rotating a model on Intel i7 /16GB is too slow to use even.... ?

markmal, 2015-06-11:
No, we draw via standard graphical functions. It is up to FPC runtime or 
X how to use acceleration.

tvld, 2015-06-11:
Hm. Then why is the rotating so much slower in Freeship than in the other software I use?... I will investigate more...

markmal, 2015-06-12:
The issue might be because 3D is completely implemented in FreeShip. All 3D stuff, like Z-buffer, and shading is implemented right in the program, using just primitive 2D graphical functions, like Canvas.Line for lines, or by setting color of pixels directly. Nothing fancy like OpenGL is used (yet).
For shading model, basically, every pixel of every surface is calculated by the program to find if it is visible, and if yes, what will be its color. And given your screen has high resolution, there are many pixels to process.
I am thinking to implement all it in OpenGL. In next major version, probably...

markmal, 2015-07-05:
I've profiled the app. Yes, it shows what I've said - For shading mode every pixel of every surface (triangle) is calculated by the program to find if it is visible, and if yes, what will be its color.
I've started moving to OpenGL direction...

markmal, 2015-12-27:
Tom,
Is performance still the issue after rendering was switched to direct memory access?

tvld, 2015-12-28:
Performance is great now Mark, thanks !

closed 2015-12-28 by tvld