markmal/freeship-plus-in-lazarus #6: Can't save files

cipman, 2015-09-21:
I installed https://github.com/markmal/freeship-plus-in-lazarus/releases/download/3.5.3.1/FreeShip-3.5.3.1_x86-64_linux_qt.zip on ubuntu 14.04/64bits (I tested the previous release as well - same thing). The program seems to work ok but I cant save anything.
When I try to save, a window pops up saying:
" Unable to create file "saved_screenshot.jpg".
Press OK to ignore and risk data corruption.
Press Cancel to kill the program."
 To reproduce:
- open Free!Ship Plus
- new file
- Save As ... (name, place etc ... click Save)
- the window with the above mentioned contents pops up.
No file saved and the program seems to continue working fine.

SergeyShorin, 2015-09-21:
Yes! I try "Save As..." function, and have same error message.
I'm use Linux Mint with MATE desktop environment.
and: if I open existing file, just change point of view (move mouse with pressed wheel) and try save - error, press cancel, but file erased, and leaves only *.bak file in project directory. 
I have project in other version of freeship. when I click Save (or "Save As"), app offer to me save in previously or current version of format. choose any - error.
try all of this issue in user interface in russian and english language.

p.s. Why app trying save *.jpg not *.fbm?

markmal, 2015-09-21:
Thank you guys,
I'll check.
 "saved_screenshot.jpg" is created for debugging/testing purposes before FreeShip saves content into .fbm. Just for me to see what picture it will write into .fbm.
Most likely you don't have write permissions in current directory where you start FreeShip.
As a temporary workaround try to open xterm and then, from your home directory, start FreeShip.

cipman, 2015-09-21:
Indeed, starting FreeShip in a terminal from any folder where the user has write permission works fine - you can save your work.
Now I see FreeShip saves a "saved_screenshot.jpg" file in the directory it was started from - so this is the source of the problem.

markmal, 2015-09-22:
The issue is fixed. Updated version is issued
FreeShip+ 4Lazarus. x86_64, Linux, Qt. Prerelease 3.5.3.21
Please test and let me know.

cipman, 2015-09-22:
The 3.5.3.21 version saves the files as it should, thanks.
But it's something wrong with it: eats up a whole CPU thread doing nothing (100% CPU single thread).
Let me know if you want me to open a new issue.

markmal, 2015-09-22:
Thanks cipman.
Yes please open separate issue for CPU looping. Please provide with more details. You could run it from xterm and copy-paste output.

This case is closed.

closed 2015-09-22 by markmal