markmal/freeship-plus-in-lazarus #7: not work Develop Plates

SergeyShorin, 2015-09-21:
Hi! FreeShip-plus-in-lazarus - it's a very good app!!! 
Thank You VERY MUCH - finally I (and more much people) can use this program on home PC with linux-OS.

But I have say few words:

first: if user use linux such as Linux Mint - explain to them, what they need install some package (qt4-dev-tools libqt4-dev libqt4-core libqt4-gui). For example Linux Mint have not QT-based desktop environment, and many people who need this program for calculation / drawing and construction of its DIY ship does not so much understand the configuration of the computer (and Linux in particular), to understand why they can not launch the program.

and second: I only try program, but already find bug - do not working "Develop Plates". I try this function on default ship (File-New) in FreeShip Plus in windows - working, in FreeShip-plus-in-lazarus - crush with "Range check error" message. After then I try this function on other free hulls from HydroShip site, and some user-made project from some internet board... and have "Range check error" message again.

p.s. Thank You very much again - porting this app to linux - very-very cool!!!

markmal, 2015-09-21:
Spasibo Sergey!

For Ubuntu just libqt4pas5 was enough. Probably the libraries requiered are in different packages in other distros.
Currently the FreeShip binary needs following Qt4 libs
        libQt4Pas.so.5 => /usr/lib/x86_64-linux-gnu/libQt4Pas.so.5
        libQtWebKit.so.4 => /usr/lib/x86_64-linux-gnu/libQtWebKit.so.4
        libQtGui.so.4 => /usr/lib/x86_64-linux-gnu/libQtGui.so.4
        libQtNetwork.so.4 => /usr/lib/x86_64-linux-gnu/libQtNetwork.so.4
        libQtCore.so.4 => /usr/lib/x86_64-linux-gnu/libQtCore.so.4
        libQtOpenGL.so.4 => /usr/lib/x86_64-linux-gnu/libQtOpenGL.so.4

You can find what packages provide this libs using dpkg-query or rpm tool like:
$ dpkg-query -S /usr/lib/x86_64-linux-gnu/libQt4Pas.so.5
libqt4pas5:amd64: /usr/lib/x86_64-linux-gnu/libQt4Pas.so.5
or
$ rpm -qf /usr/lib64/libQtCore.so.4
libqt4-4.8.6-4.4.1.x86_64

I think it will not harm to install all libQt4 related packages.

I'll check "Develop Plates" issue.

markmal, 2015-09-22:
The issue is fixed. Updated version is issued
FreeShip+ 4Lazarus. x86_64, Linux, Qt. Prerelease 3.5.3.21
Please test and let me know if it works or not. If it does not, please send me your .fbm

SergeyShorin, 2015-09-22:
Ok. i try this at home (at office - windows, at home - linux)
also I try fixed "Save As" from neighbor issue # 6.
if all work fine - I say You, what You can label 6 & 7 issue as "Closed"

About required libraries - I think You can place a list of required libraries (directly copy from Your message in this issue) in a "required library note", at download pages for non-experienced user... but this is IMHO

Thank You for quick response and don't leave users alone with their problems.

SergeyShorin, 2015-09-22:
YES! issue fixed
Thank You!

markmal, 2015-09-22:
Thank you Sergey!
closed

closed 2015-09-22 by markmal