markmal/freeship-plus-in-lazarus #8: 100% CPU

cipman, 2015-09-22:
I deleted all FreeShip ini/config files from old version (not required though - I just simulated a fresh install).
I installed https://github.com/markmal/freeship-plus-in-lazarus/releases/download/3.5.3.21/FreeShip-3.5.3.21_x86-64_linux_qt.zip system wide on ubuntu 14.04 /64.
As long as there's no file opened (even new) FreeShip seems to work fine. If you open a file (or use a new one with defaults), FreeShip v3.5.3.21 uses 100% CPU.

How to reproduce:
- start FreeShip (from terminal or Unity etc - doesn't matter)
- if there was no filed previously saved (or no ini file) the FreeShip screen will be blank (no viewports) - all is good till now, FreeShip doesn't use CPU on idle (check with System Monitor or alike)
- File > New > Ok (make a new file/project and accept the defaults)
- FreeShip uses 100% CPU - thats's not good.

The output when starting FreeShip from terminal (with my comments preceded by ***):

*** Program start 
Compiled at 2015/09/21 23:05:42
Compiler version: 2.6.4
Target CPU: x86_64
Target OS: Linux
FreeShip Product version: 3.5.0.0
FreeShip Program version: 3.5.3.21
Last SVN Change Revision: 31
*** All OK till here
*** File > New > OK
*** This comes on terminal:
SaveBinary
Stations:0 pos:0
Buttocks:0 pos:0
Waterlines:0 pos:0
Diagonals:0 pos:0
Markers:0 pos:0
ResistanceDelftData: pos:0
ResistanceKaperData: pos:0
backgroundImages:0 pos:0
Flowlines:0 pos:0
TFreeViewport.Create: done
TFreeViewport.Create: done
TFreeViewport.Create: done
TFreeViewport.Create: done
*** Now FreeShip uses 100% CPU 

If there was a previously saved file, FreeShip starts with that file and uses 100% CPU from the start.

cipman, 2015-09-22:
P.S.
The previous prerelease 3.5.3.1 doesn't have this problem (I just checked).

SergeyShorin, 2015-09-22:
1. I just test 3.5.3.21, and have not 100% CPU load... may be a little - 10-20% before start FreeShip, and 32-37% when FS running (I not do any in FS - just open and load project)

but see new problem:

2. starting program goes a little bit slowly. after I click icon - first drawing transparent-frame with little shaded (size not exact size of application main window), after then - redrawing main window

and

3. open last opened project. I suggest what this is "not-commented" function in start procedure (I'm VB6 programmer, and don't know name of this procedure in lazarus or in pascal... in my language I use this for debugging purpose, such quick compile and run program without the need to give the command "open the project or file" in my application)

so

may be paragraph 1 is derived from paragraph 2, which (in turn) derives from paragraph 3 - when start, FreeShip do routine for open last project after make but before draw (or refresh or paint... etc) mainwindow, and CPU take overhead-load

markmal, 2015-09-22:
Thank you guys, I'll check. May be it is some artifacts from OpenGL. I started development of OpenGL based visulaization. It is normally disabled now. But code is still there in application. May be it somehow interacts with Unity/Compiz. I have XFCE desktop on my Ubuntu, not Unity. May be that is why I have not noticed this CPU load.
Sergey, what desktop do you use?

SergeyShorin, 2015-09-22:
I use linux mint with MATE desktop (like old gnome DE)
You can see: http://www.linuxmint.com/rel_rafaela_mate.php

markmal, 2015-09-23:
Yes, I see 100% CPU in my XFCE too. 
Caused by constant window redraw, like WMPAINT message comes again and again nonstop.
I am in process of investigation ...

SergeyShorin, 2015-09-23:
Yes! transparent window on startup looks like problem with some (one or more) window redraw.
and when running - slow redrawing simple project (created on "New" submenu) even when I just rotate, pan or zoom model in any window (top, deck, etc).

markmal, 2015-09-23:
No :) window on startup is splash window. Not sure why it is transparent in your case. In my case it shows picture. Does MATE support translucent windows?

The actual issue was in original code that worked well previously (though not very efficient). This code draws grid with sizes. 
If you turn it off in 3.5.3.21, CPU will drop down to 0%.
The code is replaced by more efficient call.

Build 3.5.3.24 where Issue #8 is fixed has been released. 
(you can download just zipped FreeShip executable). 

Please test and let me know.

cipman, 2015-09-23:
Version 3.5.3.24 seems to work fine, no more 100% CPU and no other glitches observed at first glance. I will put it to work over the w/end.
Thank you!

SergeyShorin, 2015-09-23:
Hi!
I'm download version 3.5.3.24.
CPU load 100% - I'm before not have this, and now do not see this.
and my problems not resolved - flash-screen without any content (even without frame border - only shade-like transparent window, and a little bit slow speed of rotating model.

do not hurry up - do everything qualitatively - wait until "after the week-end"

markmal, 2015-09-23:
Hi Sergey,
I understand that the issue #8 "CPU load 100%" is resolved. I will wait until next week though, before close it.

Please open separate issue about shade-like transparent splash window.

There is already issue #5 "Shading model too slow" about slow speed of rotating model. Please check if in your case symptoms are similar.
Unfortunately it is not simple to resolve it. Currently all 3-d rendering is programmed, literally color of every point of 3-d picture of shaded rotated model is calculated. That is why it is slow, especially on HD/QHD screens. I've started to convert graphics to OpenGL, but I do not know when it will be done.
What screen sizes do you have?

SergeyShorin, 2015-09-24:
screen resolution 1366x768 (see next issue)

closed 2015-09-30 by markmal

cipman, 2015-10-01:
I'm afraid it's still an issue when opening the Linesplan window. There's a continuous redraw that takes 100%CPU again.
How to reproduce:
- open FreeShip > New (or keep the existing project) 
- menu > Project > Linesplan 
=> the new linesplan window pops up => 100%CPU and you can see the cursor rapidly changing from cross to wait mode; close the linesplan window and all ok.

markmal, 2015-10-01:
Hi cipman, thank you for finding it.
However this one is in different place of the program.
Could you please open new separate issue?

cipman, 2015-10-01:
Thanks for fixing it :)
Sure, no probs. => https://github.com/markmal/freeship-plus-in-lazarus/issues/13