markmal/freeship-plus-in-lazarus #9: shade-like transparent splash window.

SergeyShorin, 2015-09-24:
when start program happens shade-like transparent splash window.
Linux Mint 17.2 Mate edition, 64 bit, Notebook HP Pavilion dv6, screen resolution 1366x768.

markmal, 2015-09-24:
I've checked in Mint 17.1 64-bit, updated, in VM VIrtualBox in Win7. It shows the splash screen fine.
I'll try to upgrade to Mint 17.2 and check again.

please privide with output of following commands:
cat /etc/lsb-release
cat /etc/os-release
uname -a

markmal, 2015-09-24:
Yes, I am able to reproduce the issue after upgrade to Mint 17.2.
It uses compositing window manager, that FreeShip splash window has problems to work with, probably.
You can work it around by disabling compositing in Control Center -> Desktop Settings -> Windows.
If you turn it off, you will see the splash window.
The splash window was designed to be semitransparent in Windows, with translucency going gradually from completely translucent to opaque. In Linux with compositing windows it was simply opaque previously. However here in Mint 17.2 it became completely translucent. 
May be there is something wrong there in code of the splash, I'll check...

SergeyShorin, 2015-09-25:
command output:

lsb-release:
DISTRIB_ID=LinuxMint
DISTRIB_RELEASE=17.2
DISTRIB_CODENAME=rafaela
DISTRIB_DESCRIPTION="Linux Mint 17.2 Rafaela"

os-release:
NAME="Ubuntu"
VERSION="14.04.2 LTS, Trusty Tahr"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 14.04.2 LTS"
VERSION_ID="14.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"

uname:
Linux notebook 3.16.0-38-generic #52~14.04.1-Ubuntu SMP Fri May 8 09:43:57 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

markmal, 2015-09-25:
This (and another one) issue is fixed. Now it should work with compositing window managers.
I've tested in Mint VM with MATE(marco and compiz), Cinnamon(compiz) and KDE (KWin).
Build 3.5.3.27 is released. You can download just the program FreeShip-3.5.327.zip.

Please test and let me know.

SergeyShorin, 2015-09-29:
sorry, I'm busy last few days.
Yes? splash window issue is closed.
Would You like to I open next some issues:
1. when start FreeShip loaded (opens) last project. If I rename file of last project (when FreeShip closed, of course) and start app - nothing loaded, opened empty main window (without any project). it's not problem - vice versa - it usually fine option, but (I think) You should make checkbox "Load last project on startup" in option window... and then that not a problem, but fine stuff! ;-)
2. I'm load file after editing it Windows version. If layers name in that file contains not-english characters - they displayed as "???????". I can rename layer use russian characters, save and load file - all ok, displayed like I typed (mix english and russian characters). tomorrow I check this file at my office computer (Win XP), and say what happened with layers names.
note: when I start FreeShip with empty MDI window (without load last project - see point 1 of this issue) I see russian name of first layer - and this is ok, but I can add some next layer (without create new project even)... ok - it funny, but lets see next algorithm:
a) add one or some layer
b) click "new", say "Yes, I want save project first"
c) have crashed application - and this is not funny.

please disable "add layer" button (may be all button on toolbar, exept "new", "open", "exit" and so on [like "help", "about"] - that not affected on existing project) until user load project or create new.

Damn! so many chars... 
ok. if I open two new issues - would You like what I move points of this message to new issues (cut here - paste there)

p.s. I try add project file, but can't. so I rename -001.fbm to -001.fbm.png - so You can see: first and second layers named in mix-language characters:
"first-первый" and "second-второй"

[image]

closed 2015-09-29 by SergeyShorin

markmal, 2015-09-29:
Sergey, thanks for working on it. Nice to know that splash window issue is tested positively. This issue is closed.
Please open other issues as new separate issues. Please do not describe them here in a comment of this issue if they are not relevant to this issue. Each issue should be tracked in a separate thread.