markmal/freeship-plus-in-lazarus #10: Open last project on startup

SergeyShorin, 2015-09-29:
when start FreeShip loaded (opens) last project. If I rename file of last project (when FreeShip closed, of course) and start app - nothing loaded, opened empty main window (without any project). it's not problem - vice versa - it usually fine option, but (I think) You should make checkbox "Load last project on startup" in option window... and then that not a problem, but fine stuff! ;-)

markmal, 2015-09-29:
FreeShip "remembers" last project by its file name. If you changed the file name outside of the program, FreeShip of course does not know new name and cannot open it. 

The checkbox "Load last project on startup". 
So if it is unchecked, FreeShip will always open empty. Is it what do you mean?
It can be added, though, if it is checked, and you rename file when FreeShip closed, FreeShip will load nothing again. :)

I assume this checkbox is a nice feature and is low priority.

SergeyShorin, 2015-09-30:
Yes! it is VERY low priority.
let me explain what I see this option:
Option Windows have two checkbox: 1-"Load last project" and 2-"ask me for load last project"
a) 1-on, 2-off -> FreeShip starting and load last project *
b) 1-on, 2-on -> FreeShip starting and ask me. if I answer "yes" - do "a)". if answer is "no" - nothing load 
b1) may be add one button "no, create new project": yes - appear "Create new" dialog, no - nothing load **
c) 1-off, 2-disabled -> nothing load, but may be "b1)"

* if file renamed outside FreeShip (or deleted) - assumed 1-off
** may be add one more checkbox 3-"ask for create new if not load last project"

User need "load nothing" functionality for reason: import data from another format to create new clear project

SergeyShorin, 2015-09-30:
if You wish - I can draw diagram for this algoritmh

markmal, 2015-09-30:
Hi Sergey, 
I am not sure if I getting a practical benefit for this new feature. 
Most of people work on one project for many days. Each time they open same project to add/modify something there. 
Why do you think adding this feature will add value for us?
May be it will make sense to add this logic (that you describe) only if the last file was not found, renamed outside, deleted?
Then logic should be:
1 FreeShip checks if last projects file exists
2. If yes, it loads it. Done.
3. If not, it brings a dialog where it tells that the file "my last project.fmb" was not found, would you like:
3.a to find it yourself
3.b to create new project
3.c to proceed with empty project (may be you want to import something later)

if user chooses: 
 3.a it will bring up "Open" dialog. 
 3.b it will bring up "New" dialog
 3.c just proceeds with empty project

How about this?

SergeyShorin, 2015-10-05:
Yes! Your offers is better! Fine algorithm!

markmal, 2015-12-21:
It works in my tests. I am closing this issue.

closed 2015-12-21 by markmal