markmal/freeship-plus-in-lazarus #11: problem with layers name

SergeyShorin, 2015-09-29:
I'm load file after editing it Windows version. If layers name in that file contains not-english characters - they displayed as "???????". I can rename layer use russian characters, save and load file - all ok, displayed like I typed (mix english and russian characters). tomorrow I check this file at my office computer (Win XP), and say what happened with layers names.
note: when I start FreeShip with empty MDI window (without load last project - see point 1 of this issue) I see russian name of first layer - and this is ok, but I can add some next layer (without create new project even)... ok - it funny, but lets see next algorithm:
a) add one or some layer
b) click "new", say "Yes, I want save project first"
c) have crashed application - and this is not funny.

please disable "add layer" button (may be all button on toolbar, exept "new", "open", "exit" and so on [like "help", "about"] - that not affected on existing project) until user load project or create new.

p.s. I try add project file, but can't. so I rename -001.fbm to -001.fbm.png - so You can see: first and second layers named in mix-language characters:
"first-первый" and "second-второй"
[image]

markmal, 2015-09-29:
FreeShip for Lazarus stores texts in Unicode (UTF8). I suspect that Windows version stores in Windows Russian codepage 1251 (8-bit character encoding).
Problem with 8-bit character encodings it is impossible to say in what language national characters (with codes >128) were typed. Code 192 means A in Russian 1251, and something else in Arabic 1255 etc. There is no such problem with UTF8.

I will check if codepage or/and language is specified in .FBM format. If yes, it will be possible to translate from UTF8 to a Windows codepage. If not, we can only rely on Language that set in FreeShip. User have to set same in Windows and in Lazarus/Linux one.

Also, please keep in mind, despite Lazarus version was started as clone from Windows one, it is getting more and more different, for following reasons: different OS, different programming environment, and improvements and bug-fixes are going independently now. Simply speaking, it is not guaranteed, that FBM files created in one program are 100% compatible with another one. Though I am trying my best.

SergeyShorin, 2015-09-30:
O! sure! I'm understanding that different between win and lin (I'm programming in window visual basic and in linux gambas)
ok.
let's see what we have in project with renaming layers in linux:
first and second layer have @#%&&^%#$% name, other have normal readable russian name.
[image]

I'm not sure what *.fbm format have "character-set option" and may consist UTF-8 characters.
and not sure what this problem can be resolved - but You see better :-) (may be You communicate with windows-developer of FreeShip)

markmal, 2015-09-30:
It is in FreeShip for Windows, correct?
What shows FreeShip for Linux? Please post similar picture.

markmal, 2015-10-02:
Sergey, please note that you have to set correct encoding in Preferences. For Russian it should be cp1251. In Windows you also should use 1251.
Please test and let me know.

SergeyShorin, 2015-10-05:
yes, in windows locale cp1251, but in linux I use utf8 (
[image]
)
same file in linux (
[image]
)

p.s. on locale.png You can see semi-transparent terminal window :-)

markmal, 2015-10-05:
in Linux FreeShip 3.5.3.29 open File/Preferences and set FBM Encoding to cp1251. 
Then save your model as .fbm file. Then open it in Windows FreeShip.

SergeyShorin, 2015-10-06:
ok. I'll try this today at home.

markmal, 2015-12-21:
Hi Sergey, 
did not hear from you.
My tests work well. 
I am closing this issue.

closed 2015-12-21 by markmal