markmal/freeship-plus-in-lazarus #12: Creating a layer in empty project

SergeyShorin, 2015-09-30:
when I start FreeShip with empty MDI window (without load last project - see issue #10) I see name of first layer - and this is ok, but I can add some next layer (without create new project even)... ok - it funny, but lets see next algorithm:
a) add one or some layer
b) click "new", say "Yes, I want save project first"
c) have crashed application - and this is not funny.

please disable "add layer" button (may be all button on toolbar, exept "new", "open", "exit" and so on [like "help", "about"] - that not affected on existing project) until user load project or create new.

markmal, 2015-09-30:
I am in process of investigating....

markmal, 2015-09-30:
Actually when FreeShip is opened, new empty project named "new model" is kind of loaded. 
It does not have points and lines, it is empty, but it has one layer. 
User can create a model by clicking New, that brings up "New Model" wizard that creates a simple hull. 
I do not know if it makes sense to save an empty project without a hull. But lets do not remove this capability, similar to as you can save empty document in MS Word, you should be able to save empty project in FreeShip, for whatever reason.

I've fixed the bug with failure on saving. Not published yet.

SergeyShorin, 2015-10-02:
ok. check in next release :-)

markmal, 2015-12-21:
It works in my tests. I am closing this one.

closed 2015-12-21 by markmal