markmal/freeship-plus-in-lazarus #15: Access violation trying to use Lackenby transformation

cipman, 2015-10-02:
How to reproduce:
- start FreeShip
- menu > FIle > New [accept defaults] > Ok (or keep previous project)
When started from terminal, this is the output after the above commands:

	Compiled at 2015/09/22 22:51:19
	Compiler version: 2.6.4
	Target CPU: x86_64
	Target OS: Linux
	FreeShip Product version: 3.5.0.0
	FreeShip Program version: 3.5.3.24
	Last SVN Change Revision: 31
	LoadBinary
	Stations:20 pos:557
	Buttocks:6 pos:1716
	Waterlines:10 pos:2314
	Diagonals:0 pos:3122
	Markers:0 pos:3123
	FResistanceDelftData: pos:3123
	FResistanceKaperData: pos:3142
	BackgroundImages:0 pos:3153
	Flowlines:0 pos:3154
	TFreeViewport.Create: done
	TFreeViewport.Create: done
	TFreeViewport.Create: done
	TFreeViewport.Create: done

- menu > Transform > Lackenby 
=> a window pops up saying
<<< Access violation.
Press OK to ignore and risk data corruption.
Press Cancel to kill the program. >>>
Terminal output:

	TFreeViewport.Create: done
	TFreeViewport.Create: done
	SaveBinary
	Stations:20 pos:0
	Buttocks:6 pos:0
	Waterlines:10 pos:0
	Diagonals:0 pos:0
	Markers:0 pos:0
	ResistanceDelftData: pos:0
	ResistanceKaperData: pos:0
	backgroundImages:0 pos:0
	Flowlines:0 pos:0
	TFreeSubdivisionControlFace.Destroy. $33263384
	TFreeSubdivisionControlFace.Destroy. $33264376
	TFreeSubdivisionControlFace.Destroy. $33265112
	TFreeSubdivisionControlFace.Destroy. $33265848
	TFreeSubdivisionControlFace.Destroy. $33266584
	TFreeSubdivisionControlFace.Destroy. $33267352
	TFreeSubdivisionControlFace.Destroy. $33268088
	TFreeSubdivisionControlFace.Destroy. $33268696
	TFreeSubdivisionControlFace.Destroy. $33269304
	TFreeSubdivisionControlFace.Destroy. $33267128
	TFreeSubdivisionControlFace.Destroy. $33270552
	TFreeSubdivisionControlFace.Destroy. $33271288
	TFreeSubdivisionControlFace.Destroy. $33271896
	TFreeSubdivisionControlFace.Destroy. $33272760
	TFreeSubdivisionControlFace.Destroy. $33273368
	TFreeSubdivisionControlFace.Destroy. $33273976
	TFreeSubdivisionControlFace.Destroy. $33274712
	TFreeSubdivisionControlFace.Destroy. $33275320
	TFreeSubdivisionControlFace.Destroy. $33275928
	TFreeSubdivisionControlFace.Destroy. $33276536
	TApplication.HandleException Access violation
	  Stack trace:
	  $0000000001FA1358
	  $000000000085C0E2 line 82 of include/buttonglyph.inc
	  $0000000000434BA7 line 278 of ../inc/objpas.inc
	  $00000000004D217B line 153 of ../objpas/sysutils/sysutils.inc
	  $000000000085D172 line 68 of include/speedbutton.inc
	  $000000000066B75E line 512 of ../objpas/classes/compon.inc
	  $000000000066B695 line 490 of ../objpas/classes/compon.inc
	  $000000000082036F line 131 of lclclasses.pp
	  $0000000000487EC9 line 4848 of include/control.inc
	  $000000000046B0C6 line 6511 of include/wincontrol.inc
	  $000000000048B6CD line 54 of include/customcontrol.inc
	  $00000000004997CD line 315 of include/scrollingwincontrol.inc
	  $000000000049B0AD line 207 of include/customform.inc
	  $000000000066B75E line 512 of ../objpas/classes/compon.inc
	  $000000000066B695 line 490 of ../objpas/classes/compon.inc
	  $0000000000A45E50 line 16168 of Units/FreeShipUnit.pas
	  $000000000066B75E line 512 of ../objpas/classes/compon.inc

cipman, 2015-10-04:
With the new 29 build the window opens and looks ok.
There's an issue with the transformation itself but that's in the windoze version as well so we'll ignore that for a while.

markmal, 2015-10-04:
Thank you cipman, I close the issue as soon as access violation error is fixed.
If something wrong with transformation itself, please open new issue and describe.

closed 2015-10-04 by markmal