markmal/freeship-plus-in-lazarus #16: Request

koendv, 2015-12-10:
There is a Lazarus plugin which converts a Lazarus project to a Debian source package: http://prof7bit.github.io/LazPackager/ .

Is it possible to make a Debian source package for Freeship plus? 

Thanks.

speters, 2015-12-11:
I did first work regarding a Debian package in my "debian" branch (https://github.com/speters/freeship-plus-in-lazarus/tree/debian/debian), which is some commits behind master, because of lack of time (sorry).
Maybe you can use this as as starting point for your packaging.

LazPackager would only give a template for a deb package. But as the focus for packaging is an automated system (make, etc.) it would be of not much use once the build scripts are written.

markmal, 2015-12-12:
Hi Sönke,
How to merge Debian project with Freeship?

koendv, 2015-12-12:
Debian New Maintainers' Guide:

https://www.debian.org/doc/manuals/maint-guide/

markmal, 2015-12-14:
I took Sönke's deb template as a base (without real GIT merging, just copied dir from it) and I am working on .deb building. It works. 
Complexity is with adding FreeShip to menu and icons for FreeShip files.
Menu looks like is working.
However there is issue is with icons for .ftm .fbm files, that show up in some linux themes, and do not in other themes.

PS/ Yes, .deb files are convenient way of distribution. On the other side, convenience of scripts is that user can install FreeShip locally, just into the user's home, which is safer. And, of course, it works in any distro.

markmal, 2015-12-18:
deb is here https://github.com/markmal/freeship-plus-in-lazarus/releases/tag/v3.5.5.7

markmal, 2015-12-21:
Deb file is available now. I am closing this issue.

closed 2015-12-21 by markmal

koendv, 2015-12-31:
I've installed the Freeship .deb package on a clean install of debian-8.2.0-amd64. Everything works fine.

Thank you and a happy new year!

koen