markmal/freeship-plus-in-lazarus #17: Installation Manual

tvld, 2015-12-13:
This summer I worked a bit on the icons and now, with Christmas holiday, thought to pick up FreeShip again.... but, somehow I cant get it installed in Ubuntu ...   Could you point me to the installation guide? Somehow I cant find it in README.md  ....

markmal, 2015-12-14:
May be. The "standard" user location is $HOME/FreeShip may be it is hardcoded here and there.
Also, it may be that names started with dot are invisible for some tools.

markmal, 2015-12-18:
Hi Tom
install-HOWTO.txt is added

tvld, 2015-12-18:
Excellent :) Two Q:
 - where can people find the .deb file?
 - on Ubuntu a.o. one would install the .deb with  `sudo apt-get install ....`   ?   or ?

markmal, 2015-12-18:
.deb file can be downloaded from same place where .zip file is. :)
How to install it is described in install-HOWTO.txt, and also can be easily found in Internet.

Following should be done (at least) to make it available to apt-get : 
1) PPA should be created somewhere in Ubuntu Launchpad
2) source package should be created, from that Launchpad should be able to build binary packages.
Currently I do not have time for this work. If someone help... ;)

then users should add PPA to their apt repos.

tvld, 2015-12-18:
;) ... I have never created PPA myself, but I know there is one for Sailcut and FreeCAD. So if we would manage, this is becoming a great Ubuntu / Ship designer set :)

tvld, 2015-12-18:
And about the .deb location... well.. I  checked http://www.hydronship.net/download.htm , don't see it there. And the ZIP file is on https://github.com/markmal/freeship-plus-in-lazarus in top right corner.... also no .deb ... ok... maybe I am sleeping ? I do have the flu, so that might the issue  ;)

markmal, 2015-12-18:
Yes, I agree that github.com is not user friendly. It rather developer friendly. 
To download binaries you have to enter Releases page.
When you enter here https://github.com/markmal/freeship-plus-in-lazarus 
there is ribbon with kind of project "stats": N commits  N branches N releases N contributors.
Click on **__releases__** link. (Yes it is not very visible.)
It should open Releases page - https://github.com/markmal/freeship-plus-in-lazarus/releases
Here you can see number of releases in sections. From Downloads sub-sections you can download binares : .zip and .deb, and zipped sources too, if you need.

tvld, 2015-12-19:
The link in READMe makes all the difference.  :)  You might even merge the install.txt there ?  Looking good :)

markmal, 2015-12-19:
Install is in Wiki :)
https://github.com/markmal/freeship-plus-in-lazarus/wiki

markmal, 2015-12-21:
I've added also man page. One can type `man FreeShip'. :)

I am closing this issue.

closed 2015-12-21 by markmal