markmal/freeship-plus-in-lazarus #18: .deb install gives non-valid

tvld, 2015-12-19:
[image]

markmal, 2015-12-19:
Thanks, I'll check.
Interesting, I've checked the .deb with lintian, it did not show this issue.

tvld, 2015-12-19:
I downloaded again and same issue. ( Ubuntu 14.04 LTS, Core i7 )

markmal, 2015-12-19:
Ubuntu Software Center demands Installed-Size field that is not mandatory in Debian Policy.
I've added this field. 
Also made some other small fixes.
Check version 3.5.5.12 in https://github.com/markmal/freeship-plus-in-lazarus/releases/

tvld, 2015-12-19:
Installs ok now :) ...  But now I get 
```
~$ freeship
freeship: command not found
```

tvld, 2015-12-19:
I see first time the logo's  I made last summer... But they have anti-alias problem which does not make me proud... will have a look ;)

tvld, 2015-12-19:
[image]

markmal, 2015-12-20:
.pxm Logo I've converted from 32/00-freeship.png with quite simple convert command

Also I've added size 96. (just to fill gap in sequence 16 24 32 48 64 96 128)

It looks like program overrides config at $HOME/FreeShip/FreeShip.ini . I'll check.

markmal, 2015-12-20:
Probably, you have not run uninstall-user.sh. Or it was too old. 
New version does not have these artefacts.
Anyway, you can fix it manually in application File/Preferences ->Directories. 
Or you can delete .config/FreeShip/FreeShip.ini or or edit it with removing all lines from [Directories] section.

tvld, 2015-12-21:
I did indeed not uninstall anything. I just hit "upgrade" when opening the .deb with Ubuntu's Software centre. 
Selecting `/usr/share/FreeShip/Languages` worked. Thanks )

tvld, 2015-12-21:
We are mixing issues here. About the icons:  `make_freeship_menu_icon_sprite.sh` should work with 96dpi icons as well. Maybe there is a wrong background color chosen to be able to anit-alias properly given that pixels can not be semi-transp in icons....

markmal, 2015-12-21:
background color? I thought icons are in RGBA format (with Alpha channel). Aren't they?

markmal, 2015-12-21:
If there are problems with icons, can we create separate issue? 
I would close this issue as soon as .deb works well and major package managers do not complain.

tvld, 2015-12-21:
Yes. Good to close it. I will play a bit more and come back to it if needed ))

closed 2015-12-21 by markmal