markmal/freeship-plus-in-lazarus #19: Serious memory leak

tvld, 2015-12-22:
Just rotating a shaded model screams up the memory usage. Each small rotate step takes more 100MB that is never freed.  After 10 minutes we are on 10GB. ( Ubuntu 14.04, 16GB,  i74770K )

[image]

tvld, 2015-12-22:
Getting many 
```
TFreeSubdivisionControlFace.Destroy. $60998440
TFreeSubdivisionControlFace.Destroy. $61012360
TFreeSubdivisionControlFace.Destroy. $61013192
TFreeSubdivisionControlFace.Destroy. $61014264
TFreeSubdivisionControlFace.Destroy. $61014808
TFreeSubdivisionControlFace.Destroy. $61015480
TFreeSubdivisionControlFace.Destroy. $61016088
TFreeSubdivisionControlFace.Destroy. $61016568
TFreeSubdivisionControlFace.Destroy. $61017048
TFreeSubdivisionControlFace.Destroy. $61017544
TFreeSubdivisionControlFace.Destroy. $61017976
TFreeSubdivisionControlFace.Destroy. $61018600
```

markmal, 2015-12-22:
I'll check memory.

TFreeSubdivisionControlFace.Destroy. Is logging.

Slowness we discussed in issue  Shading model too slow #5 
I am working on moving graphics to OpenGL...

markmal, 2015-12-22:
Hmmm. heaptrc is clean
Heap dump by heaptrc unit
0 memory blocks allocated : 0/0
0 memory blocks freed     : 0/0
0 unfreed memory blocks : 0
True heap size : 0
True free heap : 0

tracing with valgrind...

markmal, 2015-12-22:
it looks like I've found the offender:
QT widget creates pen object each time when Canvas.SetPixel called from program.
that happens tons of times especially in shading mode.
Working it around will require to use direct work with bitmap, a format of that can be different in different environments. Some have RGB, some RGBA, some ARGB etc...
One more + for OpenGL.
Also I'll try to create a little example of Canvas.SetPixel leak and post on Lazarus QT widget bug log.

tvld, 2015-12-22:
Smart find :)  Seems FS uses the QT widget in an unusual way? Even if QT fixes the issue quickly, the patched version will not be distributed quickly over various distributions?

Indeed +1 OpenGL.

markmal, 2015-12-23:
Not FS. Lazarus does. 
I've switched from Canvas.SetPixel method to direct memory access. Works fast, and I do not see leaks for now. However it may fail on exotic screen formats, especially low color low bit.  With palettes etc.
Only main ones are supported like 24 bit and 32 bit ones by this method.

tvld, 2015-12-23:
Ok, I am curious. Please let me know when I can update with a new .deb   :+1:

markmal, 2015-12-23:
3.5.5.19 is issued

tvld, 2015-12-24:
You released 3.5.5.19  5 days ago, before your fix ? 
Anyway, I installed and indeed good news:  hige performance increase and memory stays nicely at 81MB. Well done, thank you :)  
Small detail: indeed colors are mixed up completely:

[image]

markmal, 2015-12-24:
With release "5 days ago" I do not know... Some issue with GitHub.

Colors I'll check. Probably RGB taken as BGR...

markmal, 2015-12-24:
3.5.5.21 issued.

tvld, 2015-12-25:
Looking good :) Thank you.

closed 2015-12-25 by tvld