markmal/freeship-plus-in-lazarus #22: Строевая по шпангоутам

vla2849, 2016-07-15:
Марк привет!

Что-то не заходишь на форум FS-па. Я обновился до последней версии Debian 8.5. Не отрисовывается строевая по шпангоутам и не отображаются характерные точки - ЦВ, ЦТ и т.д.

С уважением,
Владимир.

markmal, 2016-10-31:
Привет Владимир!
Все лето отдыхал, занимался практическим судостроением (делал и ремонтировал гидрофойл для виндсерфинга, учился кататься)
Скоро посмотрю в чем проблема.

vla2849, 2016-11-02:
Добрый день Марк! У меня подозрение на то, что в восьмом дебиане по умолчанию система инициализации systemd, может где-то что-то не докрутили. И вообще с этой системой инициализации стало не возможно нормально выключить ноутбук, т.е. из меню (у меня LXDE). В предыдущей 7-й всё работало отлично. Пытаюсь разобраться, но так как с английским напряг, то приходится искать на русском. А с подводным крылом к соревнованиям допускают? Интересно, при какой скорости ветра виндсёрфинг выходит на крыло? А если наедешь на подводное препятствие - крыло на отвалится? -- С уважением,               Владимир Ветров.

vla2849, 2016-12-31:
Марк привет!
Поздравляю с наступающим Новым 2017 годом! Желаю успехов как в работе, так и в жизни и самого главного - богатырского здоровья для занятий парусным спортом, ну и вообще быть здоровым всегда и везде.

Ну а теперь немного о грустном. Я специально поставил FS под WIN 7. чтобы посмотреть будет ли отрисовываться строевая по шпангоутам и характерные точки - увы этого не случилось. 
[image]

Ещё при выходе из программы появилось сообщение
[image]
Жмёшь "OK" - из программы не выходит, "Cancel" - выходит. Как говорится - "вот такие новогдние пироги".

markmal, 2019-05-22:
Issue with slashes / \ is fixed.

closed 2019-05-22 by markmal