markmal/freeship-plus-in-lazarus #23: http://hydronship.net EXPIRED???

Symbian9, 2016-10-26:
Where official site now?
* http://www.hydronship.net/

Currently it give this
> This domain name expired on Oct 15 2016 06:33:22:000AM

* Snapshot - http://archive.is/crHu7

markmal, 2016-10-31:
Hi!
FreeShip Plus in Lazarus is independent fork of FreeShip Plus from Hydronship. 
I do not know of destiny of the former one and do not control it.
hydronship.net is not official site of FreeShip Plus in Lazarus. It looks like some generic domain name search site. 
It is sad to know that Hydronship is not supported any more. At least as it looks like.

FreeShip Plus in Lazarus is advanced with fixing multiple bugs, as well as some improvements have been implemented.
It was started with thought to be able to work in Linux.
However the latest version is compiled for Windows too.

Symbian9, 2016-10-31:
But here is actual official mirror for original FREE!Ship Plus by Timoschenko V.F.
* http://freeship-plus.ho.ua/

So, I recommend you use this mirror link!

Symbian9, 2016-11-22:
http://freeship-plus.ho.ua/ now return Error 403...

WTF?

gardner, 2017-10-12:
I just called the domain squatters. They want $4,888.00USD for the domain.

closed 2019-05-22 by markmal