markmal/freeship-plus-in-lazarus #26: dll error

helicomarcelo, 2016-12-19:
After installation it appears the missing dll information "libstdc ++ = 6.dll" and libgcc_s_dw2-1.dll "

markmal, 2016-12-20:
Hi,

I have not tested the program with Win 10.

helicomarcelo, 2016-12-20:
Do you know if it is installer or windows error.

markmal, 2016-12-20:
It is Windows and application compatibility error.
The application depends on DLL that is absent in Windows.

helicomarcelo, 2016-12-21:
I have managed to solve the problem with the dll, but now the error is (0xc000007b).
You know of someone who has managed to install on windows.

markmal, 2016-12-21:
I've installed and tested on Win 7

I'll check if I could find Win 8 and 10

Please send full error message

helicomarcelo, 2016-12-21:
The message that appears is " The application was unable to start correctly (0xc000007b). Click Ok to close the application".
This happens in win 7 e win 10.

vla2849, 2016-12-29:
Марк привет!
Вот такая ошибка в ОС WIN7. Где же взять эту dll-ку?
[image]

markmal, 2016-12-29:
> error-fs-3 6

It should be in QT4 Runtime. Should be installed separately.
What QT4 is installed on your Win7?

Эта DLL должна быть в QT4 рантайме. Его надо установить отдельно.
Какой QT4 установлен на Win7?

vla2849, 2016-12-30:
У меня под Win вообще не установлена QT4 рантайм. Может необходимые DLL-ки в составе релиза загружать? Пошарил по сайту QT нашёл только это -http://download.qt.io/official_releases/qt/4.8/4.8.7/qt-opensource-windows-x86-vs2010-4.8.7.exe и http://download.qt.io/official_releases/qt/4.8/4.8.7/qt-opensource-windows-x86-mingw482-4.8.7.exe. Что же из них устанавливать?

vla2849, 2016-12-30:
Ещё нашёл утилиту Dependency, которая показывает чего же не хватает программе для выполнения в среде ОС. На перво скрине чего не хватает. На втором скрине разные архитектуры CPU.
[image]
[image]

markmal, 2016-12-30:
Please install 
http://download.qt.io/official_releases/qt/4.8/4.8.7/qt-opensource-windows-x86-vs2010-4.8.7.exe
I cannot include Qt DLLs into FreeShip package. Qt4 runtime needs to be 
installed.

Я не могу включить Qt ДЛЛ-ки во FreeShip. Необходимо установить Qt4 runtime.

markmal, 2016-12-30:
I've read Qt instructions again. I seems that it allows including just 
required Qt DLLs into package for redistribution.
Please postpone with installation of full Qt4 opensource.

I will find all required DLLs and include them into FreeShip.

Я прочитал Qt инструкции, кажется Qt позволяет включать только 
необходимые ДЛЛки в свой пакет для редистрибуциии.
Пока не ставьте Qt4 opensource.
Я найду все требуемые ДЛЛ-ки и включу в пакет FreeShip.

markmal, 2016-12-30:
All required Qt4 DDLs are included.
You do **not** have to install Qt4-opensource.

Все необходимые ДЛЛ-ки включены в пакет.
Вам **не** надо ставить Qt4-opensource

markmal, 2019-05-22:
Starting from release 4.1.1 FreeShip does not depend on QT. 
FreeShip works in Win10

closed 2019-05-22 by markmal