markmal/freeship-plus-in-lazarus #33: Export to IGES missing a piece?

xqroychen, 2018-05-24:
I am using the Export to IGES feature in the latest windows version. It seems to me that for any input, the output IGES always missing a piece. Maybe some 0, 1 index problem on windows?

ddjokic, 2018-10-03:
You should manually patch iges file. Appearently, Umax/min and Vmax/min are missing in FreeShip iges file:
https://silentdynamics.de/en/2018/03/17/importing-iges-file-from-freeship-to-opencascade/

Symbian9, 2018-10-27:
> - https://silentdynamics.de/en/2018/03/17/importing-iges-file-from-freeship-to-opencascade/

Here is quote:

> ### Importing IGES File (*.IGS) from FreeShip to OpenCASCADE
> 
> FreeShip software is a convenient tool for designing hulls.
> 
> The capabilities of FreeShip are essentially limited to the design of the fuselage exterior. For everything else, a real CAD system is required. There is an IGES export function for the transfer.
>
> There is also a successor: DelftShip. The IGES export function has been removed from the free version of DelftShip and is only available in the commercial version. This is not available to me, so I cannot test it.
> 
> I would like to use the geometry in our tool ISCAD. It is based on OpenCASCADE.
The hull geometry modeled in FreeShip 2.6 looks like this:
> 
> [image]
> 
> The export to an IGES file is unspectacular and 40 individual faces are created.
> 
> The next step is the import into OpenCASCADE. But there is a problem here. OpenCASCADE (v7.2.0) reports:
> 
> > Report : 40 unknown entities.
> > 
> > Total number of loaded entities 41.
> 
> Nothing is displayed. Although it is reported that an import works with earlier versions  (https://forum.freecadweb.org/viewtopic.php?t=1670 ), the import does not work with the current OCC version, nor with different older versions of Salome (and OpenCASCADE). On the other hand, the import works e.g. in the commercial CAD software Creo.
> 
> Finally, a study shows that the file exported by FreeShip contains only entities of type 128 (spline surface). In addition, there is a single color definition at the beginning. At the end of the parameter block of a 128 surface (see e.g. https://wiki.eclipse.org/IGES_file_Specification#Rational_B-Spline_Surface_.28Type_128.29 ) are the start and end parameters (min/max U and V values) of the surface. These entries are omitted by FreeShip and for OpenCASCADE this is an error.
> 
> One workaround is to patch the IGES import of OpenCASCADE. The corresponding code is located from line 188 in the file IGESGeom/IGESGeom_ToolBSplineSurface.cxx. I removed the error message and inserted default parameter limits:

```	
  if (!PR.ReadReal(PR.Current(), aUmin) || !PR.ReadReal(PR.Current(), aVmin)){
    //Message_Msg Msg106("XSTEP_106");
    //PR.SendFail(Msg106);
    aUmin=0.0;
    aVmin=0.0;
  }
 
  if (!PR.ReadReal(PR.Current(), aUmax) || !PR.ReadReal(PR.Current(), aVmax)){
    //Message_Msg Msg107("XSTEP_107");
    //PR.SendFail(Msg107);
    aUmax=1.0;
    aVmax=1.0;
  }
```

> With these modifications, the import works:
> 
> [image]

markmal, 2019-05-21:
Hi, the file loads into FreeCAD successfully after I put aUmin=0.0; aVmin=0.0; aUmax=1.0;  aVmax=1.0; directly to IGES file into each section type 128.
I can just dumbly add these values to be written to end of each type 128 section. Not a big deal.
However, from the perspective of geometry, these "Start and End parameters" of Regular B-Spline surface have some meaning. I just do not understand what the meaning is and how to calculate these values correctly. Could someone help?

markmal, 2019-05-23:
After reading some literature on IGES and B-splines, some experimentation with FreeCad, my understanding is that these values define domain of knot values. And this is normalized to [0,1]. 
So, if this is correct, I will just add these values to the end of type 128 entity.

markmal, 2019-05-23:
The fix is done. If no other bugs will appear I will release new version this weekend.

markmal, 2019-05-24:
released 4.1.2

closed 2019-05-27 by markmal