markmal/freeship-plus-in-lazarus #35: Encoding or interface problems

mrkaban, 2019-04-10:
Hello, apparently this problem has already been raised in ticket 25 ([link](https://github.com/markmal/freeship-plus-in-lazarus/issues/25)), but the person did not describe the problem to the end and did not begin to deal with the problem.

When running on Windows 10 x64 and Windows 7 x64, the text in the dialog boxes is incorrect, and the icons are incorrectly displayed.

The installation script ran. The program and the script run on behalf of the administrator.

mrkaban, 2019-04-10:
[image]
[image]
I attach screenshots, when I run the program, an error is displayed, it is also on the screenshot.

markmal, 2019-05-21:
Fixed in 4.1.1.159

closed 2019-05-21 by markmal

mrkaban, 2019-05-23:
Forced again to raise the topic, since on Windows 10 x64 the same problem. version 4.1.1.159.

On Windows 7, I did not notice such a problem, so we can conclude that it was solved on this version of the OS.

[image]
[image]