markmal/freeship-plus-in-lazarus #37: Encoding or interface problems

mrkaban, 2019-05-23:
Forced again to raise the [topic](https://github.com/markmal/freeship-plus-in-lazarus/issues/35), since on Windows 10 x64 the same problem. version 4.1.1.159.

On Windows 7, I did not notice such a problem, so we can conclude that it was solved on this version of the OS.
[image]
[image]

markmal, 2019-05-24:
I see forward slash there in the path. It is weird. Some Windows tolerate forward slash in path, some do not. I'll take a look.
What is edition of your Windows?

mrkaban, 2019-05-24:
[image]

mrkaban, 2019-05-24:
On Windows 7 64-bit program started without problems. There this problem is solved.

markmal, 2019-05-24:
I've fixed the / problem, but not released the new version yet.

markmal, 2019-05-24:
released 4.1.2

mrkaban, 2019-05-25:
The same, launch error and no icons.

markmal, 2019-05-25:
fixed more "/ issues"
released 4.1.2.168
please try again

mrkaban, 2019-05-26:
The same, can you drop some log file?

markmal, 2019-05-26:
Have you run install-usr.cmd ?
Please try to run uninstall-usr.cmd, then

Installation from ZIP
1. Create temporary directory
2. Copy FreeShip*.zip file into it and unzip it.
3. Run install-user.cmd

Could you find where is Themes directory? In Win10 it is in %HOME%\AppData\Roaming\FreeShip\

mrkaban, 2019-05-27:
Yes, I just tried it. This folder is here "C:\Users\Acer\AppData\Roaming\FreeShip".

I attach the contents of this folder, suddenly help.
[FreeShip.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/3223839/FreeShip.zip)

mrkaban, 2019-05-27:
Previously deleted, then installed, and each action on behalf of the administrator performed, but still the same error.

markmal, 2019-05-28:
in your zip Themes directory is empty. It should have subdirectories "Default" and "Original" and further tree of subdirectories and icons.
I am not sure why XCOPY did not copied this tree. 
the install-user.cmd should be run under regular user, not Administrator. 
It will install application into your %HOME%\FreeShip 

I have manually removed %HOME%\FreeShip and %HOME%\AppData\Roaming\FreeShip
Then unzipped installation zip into %HOME%\tmp\FreeShip
Then just double clicked install-user.cmd as normal user.
The FreeShip was successfully installed into my home and when run it I do not see any problem.

mrkaban, 2019-05-28:
I unpacked it in C:\Users\Acer\tmp\FreeShip and launched install-user.cmd from this folder as an administrator. When starting from the same folder, the same error appears. Have you checked the program on this operating system?

markmal, 2019-05-28:
Please run install-user.cmd regular way. Do not use "Run As Administrator".

mrkaban, 2019-05-29:
That helped! Thank!

mrkaban, 2019-05-29:
Added a program to my website about free software with an attendance of approximately 1500 people per day. He is in Russian, here is the [link](https://xn--90abhbolvbbfgb9aje4m.xn--p1ai/%D0%B3%D1%80%D0%B0%D1%84%D0%B8%D0%BA%D0%B0/%D1%81%D0%B0%D0%BF%D1%80/free-ship-plus-%D0%B2-lazarus.html).

mrkaban, 2019-05-29:
Ticket can be closed! Thanks for the quick answers!

markmal, 2019-05-31:
Thanks for the reference!

closed 2019-05-31 by markmal