markmal/freeship-plus-in-lazarus #38: Freeze

pafurijaz, 2019-05-26:
Hi and thank you for the developing, I have a installed the latest build but when I execute, after splash screen and open dialog box It freeze with all OS, my system is an old Hp g4 250 with Intel Skylake and GMA 5500 my OS is an Xubuntu 18.04.2

markmal, 2019-05-26:
Hi. Is your  OS 86_64 ?

pafurijaz, 2019-05-27:
yes :)

markmal, 2019-05-28:
What is output if you run FreeShip from shell?
What is you graphical settings?

pafurijaz, 2019-05-28:
~$ FreeShip %f
FreeShip in Lazarus
Compiled at 2019/05/25 14:04:01
Compiler version: 3.0.4
Target CPU: x86_64
Target OS: Linux
FreeShip Product version: 4.1.1.0
FreeShip Program version: 4.1.2.166
Last Git Change Revision: 202

(FreeShip:4048): GLib-GIO-CRITICAL **: 17:18:20.600: g_dbus_proxy_new: assertion 'G_IS_DBUS_CONNECTION (connection)' failed
WARNING: TResourceCacheItem.IncreaseRefCount 1000 TFontHandleCache

pafurijaz, 2019-05-28:
The graphical setting is the default of distro, with opensource driver for intel GMA 5500

pafurijaz, 2019-05-28:
I tried with a better test, after after an endless wait for the thumbnail preview creation (I have a lot o hull), I had success on open a hull, after, and now is very fast in view port and I have some other goodness..

markmal, 2019-05-28:
happy to know all worked out well. Yes for very complex hulls it may load preview for looong time. 
When you save it will save preview as bitmap and next time will load fast.  Make sure you have checked Save Preview Image in Project Settings.

I'll think, may be to limit "live preview" option somehow....

closed 2019-05-28 by markmal

pafurijaz, 2019-05-29:
Ok you closed this but I think there is anyway something that doesn't work well, and when I figure it bug out I create a better reporting of this issue, and and mid time I hope you fix this lacks because freeze my pc when i try to start the application from menù