markmal/freeship-plus-in-lazarus #39: open file crash linux

pafurijaz, 2019-06-01:
Hi the latest build on my os (xubuntu 18.4) crash when I try to open a file.
[image]

markmal, 2019-06-03:
Hi, I see the project is open. It crashes after that. 
Try to run FreeShip from bash terminal, what output do you see after crash?
If you could share (or email) the file it would be much simpler for me to trace.

pafurijaz, 2019-06-03:
works well if I execute from terminal with FreeShip %f and I get this messages:

 (FreeShip:8287): GLib-GIO-CRITICAL **: 20:02:11.414: g_dbus_proxy_new: assertion 'G_IS_DBUS_CONNECTION (connection)' failed
WARNING: TResourceCacheItem.IncreaseRefCount 1000 TFontHandleCache

markmal, 2019-06-04:
I was able to reproduce a bug. Not sure if same. 
I created new hull with 60x50 points. Started program without --nosplash parameter. During loading I clicke to splash window to close it. After that I got the error.
There is code that tries to access objects of splash after it destroyed. I will fix it and let you know.
Workaround: run with --nosplash
./FreeShip --nosplash
PS/ using --nosplash you still agree with legal conditions of program usage. "No warranty of any kind...etc...etc..."

pafurijaz, 2019-06-04:
Thanks for the tips, and for the developing. Mine Friend has this issue me too with wine. "It opens and seems to work in Windows 10, but all buttons across the top are just black boxes."

markmal, 2019-06-04:
for Windows please follow install-HOWTO.txt

pafurijaz, 2019-06-05:
Thanks man!

commit e7c9f676 (2019-06-06): bugs fixed: #39. New: Show stack on exception. Saving/loading of groups - v4.3
closed 2019-06-06 by markmal