markmal/freeship-plus-in-lazarus #41: Impossible load some old files.

pafurijaz, 2019-06-07:
Hi in the testing I had a problem to load some oldest files here some heavy hulls made by me..

[boat_hull.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/3267117/boat_hull.zip)

markmal, 2019-06-08:
I was able to load all files. Some of them, like yacht, it took >10min to open. Just be patient.

pafurijaz, 2019-06-08:
Ok but with the old version they are loaded very fast.. and the version of FreeShip under wine on linux load fastest.

markmal, 2019-06-08:
May be. I've turned on all checks in compiler. For types, arrays etc. Of course it slows everything down. When the program will be more or less stable I will turn off the checks.

pafurijaz, 2019-06-09:
OK thanks I will wait for the next releases for now thanks. And now you have a very heavy file to test :)

closed 2019-06-09 by pafurijaz