markmal/freeship-plus-in-lazarus #43: wrml import file crashes

pafurijaz, 2019-06-23:
Hi i tried to import a file from vrml 1 but I had an instantly crashes here the terminal output.

`(FreeShip:11824): GLib-GIO-CRITICAL **: 09:57:27.176: g_dbus_proxy_new: assertion 'G_IS_DBUS_CONNECTION (connection)' failed
WARNING: TMenuItem.Destroy with LCLRefCount>0. Hint: Maybe the component is processing an event?
Runtime error 219 at $00000000004223D0
  $00000000004223D0
  $0000000000A0F07F
  $00000000004D7479
  $0000000000632FDC
  $000000000069ED3D
  $0000000000632CBF
  $000000000069826A
  $0000000000432C1F
  $0000000000864175
  $0000000000906F21
`
below the file:
[Vaporetto_Venezia_01.wrl.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/3317595/Vaporetto_Venezia_01.wrl.zip)

markmal, 2019-06-23:
Hi, I'll check

markmal, 2019-06-25:
Fixed. Will release soon.

markmal, 2019-07-01:
4.3.2 released

pafurijaz, 2019-07-03:
now works :)

closed 2019-07-03 by pafurijaz