markmal/freeship-plus-in-lazarus #46: libgdk-x11-2.0.so.0: cannot open shared object file

Symbian9, 2019-07-31:
I just tried to install latest version:

1. Download and install [`freeship_4.3.2.195-all_amd64.deb`](https://github.com/markmal/freeship-plus-in-lazarus/releases/download/4.3.2/freeship_4.3.2.195-all_amd64.deb);
2.  Replaced `/usr/bin/FreeShip` with [`FreeShip`](https://github.com/markmal/freeship-plus-in-lazarus/releases/download/v4.4-alpha.1/FreeShip) asset from [`v4.4-alpha.1`]https://github.com/markmal/freeship-plus-in-lazarus/tree/v4.4-alpha.1) release;
3. Tried to launch it but got warning of lost dependency:

```
mypc@mx:~
$ FreeShip
FreeShip: error while loading shared libraries: libgdk-x11-2.0.so.0: cannot open shared object file: No such file or directory
```

**OS**: MX Linux MX-17.1 (Debian 9.x), x86_64

markmal, 2019-08-01:
please run  	required-packages.sh
install packages after "Required packages that are not installed" line

Symbian9, 2019-08-01:
> please run `required-packages.sh`

But, I use DEB:

- https://github.com/markmal/freeship-plus-in-lazarus/releases/download/4.3.2/freeship_4.3.2.195-all_amd64.deb

Should I get ZIP too?

- https://github.com/markmal/freeship-plus-in-lazarus/releases/download/4.3.2/FreeShip-4.3.2.195_x86-64_linux_gtk2.zip

BTW, it would be cool if you will setup *Travis-CI* for continuous build 'nightly' *AppImage* (portable package for Linux)

- https://github.com/appimage/AppImageKit

markmal, 2019-08-02:
I do not think you have to get ZIP. Just try to download required-packages.sh from here.

Symbian9, 2019-08-04:
>  do not think you have to get ZIP. Just try to download `required-packages.sh` from here.

OK, I has installed [](https://github.com/markmal/freeship-plus-in-lazarus/releases/download/4.3.2/freeship_4.3.2.195-all_amd64.deb), where original `FreeShip` file replaced with [`FreeShip`](https://github.com/markmal/freeship-plus-in-lazarus/releases/download/v4.4-alpha.1/FreeShip) file. So file structure look next:

```
/usr/
├ /bin/
│  └ FreeShip
├ /etc/
│  └ /FreeShip/
│      └ FreeShip.ini
└ /share/
    ├ /FreeShip/
    └ ...
```

> please run `required-packages.sh`
>
> install packages after "Required packages that are not installed" line

I just download [`required-packages.sh`](https://github.com/markmal/freeship-plus-in-lazarus/blob/master/required-packages.sh) (placed in `/home/mypc/Downloads` folder). Here is terminal output:

```
mypc@mx:~/Documents
$ ./required-packages.sh
readelf: Error: 'FreeShip': No such file
All required packages are installed
```

markmal, 2019-08-09:
you should run ./required-packages.sh ether in a folder where FreeShip is, or copy FreeShip binary into a folder where ./required-packages.sh is. Or edit ./required-packages.sh and put full path to FreeShip there.

Symbian9, 2019-08-10:
> Or edit `./required-packages.sh` and put full path to `FreeShip` there.

I choose this way.

Here is output:

```
mypc@mx:~/Documents
$ ./required-packages.sh
All packages that are required: libatk1.0-0 libc6 libc6-i386 libc6-x32 libcairo2 libgdk-pixbuf2.0-0 libglib2.0-0 libgtk2.0-0 libmagic1 libpango-1.0-0 libpangocairo-1.0-0 libx11-6
Required packages that are not installed:
$
```

So, it say that all dependency installed!

markmal, 2019-08-11:
the file FreeShip I've placed separately in release "FreeShip+ 4Lazarus. Linux-x86_64, Windows-86_64, 4.4-Alpha" is 32-bit. 
I've modified required-packages.sh so it shows architecture as well.
If your Linux is 64 bit you better install deb file.

closed 2019-08-11 by Symbian9