markmal/freeship-plus-in-lazarus #47: Latest install for Linux?

tvld, 2019-08-06:
What would the best way to install latest Linux / Ubuntu version? The .deb is no longer here?

( Visiting back after a long time.. Nice to see you still use the icons I once contributed. They seem a bit out-of-date now, don't they? )

markmal, 2019-08-09:
Hi, I've released the latest version for Linux, deb and zip.

Icons are great! may be we will need more for new functions - anchoring points, permanent adjustment to a line, etc.
Your help will be greatly appreciated!

tvld, 2019-08-09:
Super Mark, thank you. 

If you can give me a list of the icons you need, I can have a look one of these days.

markmal, 2019-08-10:
These features are still experimental. May be later.

tvld, 2019-08-10:
ok,Let me know ;-)

closed 2019-08-10 by tvld