markmal/freeship-plus-in-lazarus #50: Build Freeship for Mac OS

dmitry-pozhidaev, 2019-11-18:
OK
I will try build FreeShip for MAc OS
I instal pascal, lazarus and..
Me got linking error:

`FreeShip.dpr(311,0) Error: Error while linking`

may be I dont install some libs ?
How see detailed error message?
I first time use Lazarus

markmal, 2019-11-19:
You can set LD_LIBRARY_PATH in your profile.

dmitry-pozhidaev, 2019-11-19:
So i think header was found at compile time.
It is error of linking
I set up environment var as
`echo export LD_LIBRARY_PATH="/usr/local/lib"' >> ~/.bash_profile`
And have no success still.

markmal, 2019-11-20:
Did you try to configure project in Lazarus? 
menu Project -> Project options -> Compiler Options - > Paths -> Libraties (-Fl):

dmitry-pozhidaev, 2019-11-20:
I write /usr/local/lib and not linking still

```
ld: library not found for -lmagic
An error occurred while linking
```

dmitry-pozhidaev, 2019-11-24:
Марк, мне сложно изъясняться на английском и это будет криво. С твоего позволения я буду на русском писать и после завершения эпопеи со сборкой для заинтересованных можно будет кратко создать отчет о проделанной работе уже на английском.
Удалось собрать проект под маком. Спасибо форуму Лазаруса - https://forum.lazarus.freepascal.org/index.php?action=post;topic=47522.0;last_msg=340411

При запуске отображается заставка. 
Открывается главное окно и диалог с первичными данными.
При нажатии на кнопку ОК генерируется исключение - картинку со скриншотом диалогового окна 
[image]
прилагаю.

Опыта в программировании под как под Лазарусом, так и под маком я не имею - я линуксовод кутишник, поэтому очень уповаю на твою помощь в доводке до рабочего состояния Фришипа под Мак.

markmal, 2019-12-15:
Hi Dmitry, 
have you fixed bitmap format issue. If so, please describe, I will add the change into trunk.

closed 2020-02-22 by markmal