markmal/freeship-plus-in-lazarus #51: Release packages depend on libQtPas4, which is not readily accessible.

sjjf, 2019-12-15:
I'm running into this problem on Ubuntu 18.04 (the current LTS release), so it's not exactly bleeding edge. I can't find any Lazarus/lcl packages that would supply the Qt4 libraries - there are packages for Lazarus 1.8 and 2.0, but they're all built against Qt5.

Would it be possible to build your releases against Qt5 instead?

markmal, 2019-12-15:
Hi. Starting from release 4.1.1.159 there is no dependency on any Qt.
Qt was used for multi-document main window only. 
Currently MDI is custom coded in the program.

sjjf, 2019-12-22:
Ah, I just realised I had an old build sitting around confusing things - after clearing out the old one it's not failing to find the QT4 libraries.

closed 2019-12-22 by sjjf