markmal/freeship-plus-in-lazarus #53: Can't figure out how to install it on Ubuntu 18.04 LTS

marianov1, 2020-01-27:
I used to work a lot with FreeShip on Windows, but I just changed it to Linux and everything is still new to me here.
I really can't figure out how to install it properly on Ubuntu.
I downloaded the .zip and .deb files from [https://github.com/markmal/freeship-plus-in-lazarus/releases/tag/4.5](url).
I run the .deb file with sudo on the terminal, and installation looks like it goes ok, and I even got the FreeShip icon to open the software on 'Aplications'. But whe I run it, the software opens and I get the message:
> You run FreeShip from installation directory.
> FreeShip is not installed.
> It will work but it may experince issues with finding various files and directories.
> Please exit and install FreeShip.
> There are two installation options:
> 1 - run install-user.sh to install it into your user home
> 2 - run install-system.sh to install it into system. It requires Administrator privileges.

I then tried extracting all the files from the .zip file, and run on terminal under sudo the 'install-system.sh':
>:~$ sudo '/home/marianov1/Downloads/FreeShip/install-system.sh' 
>[sudo] password for marianov1: 
>Installation FreeShip into the system
>  Install executables
>cp: -r not specified; omitting directory 'FreeShip'
>ERROR: You do not have privileges to install FreeShip on system level.
>Try to execute with sudo or as root.

and 'install-user.sh':
>:~$ sudo '/home/marianov1/Downloads/FreeShip/install-user.sh' 

> Installation FreeShip into user home
>   Install executables
> cp: -r not specified; omitting directory 'FreeShip'
> chmod: cannot access '/home/marianov1/bin/FreeShip': No such file or directory
> cp: cannot stat 'uninstall-user.sh': No such file or directory
> cp: cannot stat 'install': No such file or directory
> cp: cannot stat 'Languages': No such file or directory
> cp: cannot stat 'Manuals': No such file or directory
> cp: cannot stat 'Ships': No such file or directory
> cp: cannot stat 'Themes': No such file or directory
> cp: cannot stat 'install': No such file or directory
> cp: cannot stat 'GNU General Public License (GPL).txt': No such file or directory
> cp: cannot stat 'Whatsnew.txt': No such file or directory
> cp: cannot stat 'uninstall-user.sh': No such file or directory
> cp: cannot stat 'install-HOWTO.txt': No such file or directory
> cp: cannot stat 'copyright': No such file or directory
> /home/marianov1/Downloads/FreeShip/install-user.sh: line 43: cd: Ships/: No such file or directory

I really don't know how to proceed from here. Should I just use the software as it is? Will I get any issue as the message says?

Sorry my still low knowledge on Linux systems, but it would be really helpful if someone could make a step-by-step for the proper installation.

marianov1, 2020-01-28:
Ok, I think maybe I got it. I checked on FreeShip menu: file>preferences;
on "Directories" tab, several locations were pointing to a :/home/FreeShip folder (that didn't existed).
So I just created it and extracted the .zip file into it.
It's working just fine now, so I think that's probably it...

Basically:
1- run the .deb package as sudo;
2- extract .zip file into new folder :/home/FreeShip.

Somebody correct me or add anything needed, please.

markmal, 2020-01-28:
Hi marianov1,
run the .deb package as sudo should do everything that is required. It is most likely a bug. I'll take a look.

it is weird that your cp does not have -r option. If it is really the case I need to adjust install-*.sh scripts.

for now you can edit $HOME/.config/FreeShip/FreeShip.ini to set proper pathes there.

markmal, 2020-01-28:
I've fixed the issue. Please download freeship_4.5.1.256-all_amd64.deb
- Backup models you developed.
- Uninstall FreeShip
`sudo apt purge freeship`
`rm -r ~/.config/FreeShip`
`rm -r ~/FreeShip`

- Install FreeShip
`sudo dpkg -i freeship_4.5.1.256-all_amd64.deb`

closed 2020-02-01 by markmal