markmal/freeship-plus-in-lazarus #55: FreeShip Plus failed on start

Symbian9, 2020-02-22:
- **Version:** `freeship_4.6.1.263-all_amd64.deb`
- **OS:** Debian 10.x buster (x86_64)

### Expected behavior

*FreeShip Plus* should work as expected

### Actual behavior

*FreeShip Plus* failed to run, but show me two error messages

[image]

[image]

### Terminal log

```
$ FreeShip
FreeShip in Lazarus
Compiled at 2020/02/19 09:16:57
Compiler version: 3.0.4
Target CPU: x86_64
Target OS: Linux
FreeShip Product version: 4.6.1.0
FreeShip Program version: 4.6.1.263
Last Git Change Revision: 287
Program exception! 
Stacktrace:

Exception class: EObjectCheck
Message: Object reference is Nil
  $0000000000A700CF
  $0000000000A70E3C
  $0000000000A6ED1C
  $000000000052F2C5
  $00000000004BA055
  $00000000004B7EAD
  $0000000000527CA5
  $00000000004CA161
  $000000000045C053
Program exception! 
Stacktrace:

Exception class: EAccessViolation
Message: Access violation
  $0000000000527E89
  $0000000000527D03
  $00000000004CA161
  $000000000045C053
[FORMS.PP] ExceptionOccurred 
  Sender=EAccessViolation
  Exception=Access violation
  Stack trace:
  $0000000000527E89
  $000000000068C23D
  $000000000046E370
  $00000000005251E9
  $000000000047284E
```

markmal, 2020-02-22:
I was able to reproduce the issue.
fixing ...

markmal, 2020-02-22:
please do
`sudo chmod 644 /etc/FreeShip/FreeShip.ini`

Also consider to install new version 4.6.1.266 with this and other fixes

closed 2020-02-22 by Symbian9