markmal/freeship-plus-in-lazarus #56: Windows x86 (32 bit) releases is absent

AZWebDDS, 2020-03-16:
Hello Mark.
When I try to install the release for x86 it cannot be installed because there is an x64-only version
What is the latest version witch working for 32?
Thanks.
Best regards Alex.

markmal, 2020-03-17:
Hi Alex
x86 is currently not supported.
The reasons are: 
- all Intel compatible CPUs in last 10 years are 64-bit capable. 
- major OS-es in last 10 years are primarily 64-bit.
- 32-bit memory model has some limitations.
- there are very little benefits to make application 32-bit.
- I do not have much time to maintain old obsolete platforms.

There are 2 possibilities for you:
1 - install Windows (or Linux) 64-bit
2 - the source code is open. You can download it and compile FreeShip for Windows 32-bit using Lazarus. Please refer how to install Lazarus and FPC here https://www.lazarus-ide.org/

AZWebDDS, 2020-03-18:
Thanks Mark.
You're right, that makes sense.
And extra thanks for Lazarus :) ... somehow I completely forgot ... I was started with Pascal in 90s, then was Delphi, and then somehow got lost it. It must be updated ;)
Thank you.

markmal, 2020-03-22:
Yes, same story with me. started with Pascal in 90s, then it was Delphi ... etc... etc...
FreeShip was originally created in Delphi. 
Welcome to Lazarus!

but any way. Consider to upgrade to 64-bit OS. 
32-bit OS is limited to 3.4GB, max 4GB of RAM, that is not enough today even for browsing only.

closed 2020-03-25 by markmal