markmal/freeship-plus-in-lazarus #57: Xubuntu GLib-GIO-CRITICAL

pafurijaz, 2020-03-23:
Hi here a problem in my machine, below the output on console..

`FreeShip
FreeShip in Lazarus
Compiled at 2020/03/22 14:43:46
Compiler version: 3.0.4
Target CPU: x86_64
Target OS: Linux
FreeShip Product version: 4.6.3.0
FreeShip Program version: 4.7.0.328
Last Git Change Revision: 335

(FreeShip:2202): GLib-GIO-CRITICAL **: 08:57:54.735: g_dbus_proxy_new: assertion 'G_IS_DBUS_CONNECTION (connection)' failed

(FreeShip:2202): GLib-GIO-CRITICAL **: 08:57:54.736: g_dbus_proxy_new: assertion 'G_IS_DBUS_CONNECTION (connection)' failed

(FreeShip:2202): GLib-GIO-CRITICAL **: 08:57:54.736: g_dbus_proxy_new: assertion 'G_IS_DBUS_CONNECTION (connection)' failed
Program exception! 
Stacktrace:

Exception class: EAccessViolation
Message: Access violation
  $0000000000CD1D95
  $0000000000CB3BF6
  $0000000000CB59A9
  $0000000000CB5E52
  $0000000000CB65F3
  $0000000000CB6855
  $0000000000CAFFCE
  $00000000004C2AD5
  $00000000004C092D
  $00000000004C8EBB
  $0000000000536438
  $00000000004C2184
  $00000000004C2705
  $0000000000472962
  $00000000004AB39D
  $000000000049DD9C
  $00000000004C40FE
`

markmal, 2020-03-23:
what is output of `cat /etc/os-release' ?

markmal, 2020-03-23:
please check if libssl1.1 is installed. If not - install

pafurijaz, 2020-03-23:
here the output :)
```
~$ cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="Ubuntu 18.04.4 LTS"
```

pafurijaz, 2020-03-23:
They are installed both 
libssl1.1  and  libssl1.1:386

markmal, 2020-03-23:
what is output of 

ls -l /usr/lib/x86_64-linux-gnu/libssl*.so*

pafurijaz, 2020-03-24:
this:
`~$ ls -l /usr/lib/x86_64-linux-gnu/libssl*.so*

-rw-r--r-- 1 root root 328128 gen  7 18:55 /usr/lib/x86_64-linux-gnu/libssl3.so
lrwxrwxrwx 1 root root     13 nov 12 17:58 /usr/lib/x86_64-linux-gnu/libssl.so -> libssl.so.1.1
-rw-r--r-- 1 root root 426232 feb 26  2019 /usr/lib/x86_64-linux-gnu/libssl.so.1.0.0
-rw-r--r-- 1 root root 577312 nov 12 17:58 /usr/lib/x86_64-linux-gnu/libssl.so.1.1`

markmal, 2020-03-24:
I've found that Free Pascal 3.0.2 OpenSSL is not compatible with libssl.so.1.1, only with libssl.so.1.0.*
If your Linux has a package like "libssl-compat" you can install it. If not, then you will have to check for updates traditional "manual" way. I'll issue another release that will not show this error on startup.
Hopefully Free Pascal 3.2 will be released soon, it is compatible with libssl.so.1.1.

pafurijaz, 2020-03-24:
I done that and I get a new row at the end..
`~/bin/startFreeShip.sh
FreeShip in Lazarus
Compiled at 2020/03/22 14:43:46
Compiler version: 3.0.4
Target CPU: x86_64
Target OS: Linux
FreeShip Product version: 4.6.3.0
FreeShip Program version: 4.7.0.328
Last Git Change Revision: 335

(FreeShip:18495): GLib-GIO-CRITICAL **: 17:40:11.740: g_dbus_proxy_new: assertion 'G_IS_DBUS_CONNECTION (connection)' failed

(FreeShip:18495): GLib-GIO-CRITICAL **: 17:40:11.740: g_dbus_proxy_new: assertion 'G_IS_DBUS_CONNECTION (connection)' failed

(FreeShip:18495): GLib-GIO-CRITICAL **: 17:40:11.740: g_dbus_proxy_new: assertion 'G_IS_DBUS_CONNECTION (connection)' failed
Program exception! 
Stacktrace:

Exception class: EAccessViolation
Message: Access violation
  $0000000000CD1D95
  $0000000000CB3BF6
  $0000000000CB59A9
  $0000000000CB5E52
  $0000000000CB65F3
  $0000000000CB6855
  $0000000000CAFFCE
  $00000000004C2AD5
  $00000000004C092D
  $00000000004C8EBB
  $0000000000536438
  $00000000004C2184
  $00000000004C2705
  $0000000000472962
  $00000000004AB39D
  $000000000049DD9C
  $00000000004C40FE
WARNING: TSpinEdit.Destroy with LCLRefCount>0. Hint: Maybe the component is processing an event?
`

markmal, 2020-03-24:
I have installed Ubuntu 19.10. 
But I cannot reproduce your issue. In my case it works.
Please try newer version.

pafurijaz, 2020-03-25:
No i use LTS for work, thanks anyway, I will continue using the original freeship that works very well with winehq, I had a problem even with wine version and sincerely I never get that working in all releases no way to works with those releases on my machine. Thanks anyway I continue seeking here for new releases maybe for something working.

markmal, 2020-03-25:
I have three machines with Ubuntu: Ubuntu Studio 16.04.6 LTS, Ubuntu 18.04.4 LTS, and recently updated one VM from Lubuntu 18.04.4 LTS to Lubuntu 19.10.
In all 3 machines I cannot reproduce your issue. I am not sure if it is specific to Xubuntu.
I've compiled last versions of FreeShip (starting from 4.7.0.332) with full debug info, including FPC RTL and packages, so it should show error stack with names.
Could you install the last version and try?

pafurijaz, 2020-03-25:
Hi, I tried to remove the package, but still not working here a new error.. ;)
`~$ FreeShip
FreeShip in Lazarus
Compiled at 2020/03/22 14:43:46
Compiler version: 3.0.4
Target CPU: x86_64
Target OS: Linux
FreeShip Product version: 4.6.3.0
FreeShip Program version: 4.7.0.328
Last Git Change Revision: 335
Program exception! 
Stacktrace:

Exception class: EAccessViolation
Message: Access violation
  $0000000000CD1D95
  $0000000000CB3BF6
  $0000000000CB59A9
  $0000000000CB5E52
  $0000000000CB65F3
  $0000000000CB6855
  $0000000000CAFFCE
  $00000000004C2AD5
  $00000000004C092D
  $00000000004C8EBB
  $0000000000536438
  $00000000004C2184
  $00000000004C2705
  $0000000000472962
  $00000000004AB39D
  $000000000049DD9C
  $00000000004C40FE
WARNING: TSpinEdit.Destroy with LCLRefCount>0. Hint: Maybe the component is processing an event?
`

markmal, 2020-03-25:
Please upgrade FreeShip to 4.7.0.334 from https://github.com/markmal/freeship-plus-in-lazarus/releases

You see GLib-GIO-CRITICAL disappeared. This error is caused by the package fcitx-module-dbus.

The Exception class: EAccessViolation Message: Access violation is because FreeShip is not compatible with libssl1.1

WARNING: TSpinEdit.Destroy... - is secondary because of the error above FreeShip exited abruptly without cleaning its resources.

pafurijaz, 2020-03-27:
Hi I tried with the latest one,now I get a new dialog box showing everytime the EULA and I agree but seem not take it into account because popup every time I start, and one other things is that FreeShip try open the latest working file but in my case was an huge file made sometime ago and it was well managed by freeship, with all the latest build was impossible work with these files that take a lot of time, after 20 minute never ready.

pafurijaz, 2020-03-27:
Here the file
[MT_Arsalan.fbm.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/4394693/MT_Arsalan.fbm.zip)

markmal, 2020-03-27:
EULA shows every time you start the application for a reason. 
Vessels are potentially dangerous things. 
I do not want to be sued for losses, damages or death of someone.
It is just one click to accept EULA.
If you do not want even this, I can give you a hint to run 
`FreeShip --help`

Slow performance is different problem than this libssl1.1 incompatibility error.
I'll close this issue and open another one.

closed 2020-03-27 by markmal

pafurijaz, 2020-03-28:
The fact is that I haven't no problem to accept the EULA but seem it don't take into account my choise where I agree, and after closing and reopening I give again the popup and If you see here in this video you can see how works.
https://drive.google.com/open?id=1mdh9KsHU1X5R_MPmNWhnt6jPac6TdpFI