markmal/freeship-plus-in-lazarus #58: Poor performance with huge complex models

markmal, 2020-03-27:
a huge file made sometime ago and it was well managed by freeship, with all the latest build was impossible work with these files that take a lot of time, after 20 minute never ready.

markmal, 2020-03-27:
@pafurijaz
I've loaded your file with about  5500 control points, 5000 faces.
It took 18 min to load. Cause - the Precision of the project is set to High.
After I've set it to Medium it takes about a minute to load.
What time is for High precision in old version? What is old version number?

pafurijaz, 2020-03-28:
Thanks, for that, For seem is trying to reload the model for every windows on splitted view, in fact I noticed that every windows take the same time and seem doing the same operation before show the hull surfaces.

markmal, 2020-03-28:
I'll check this, but it is unlikely. Please read documentation about how subdivision works.
It will give you understanding why precision has so much impact on performance.

Currently I am profiling the program to find bottlenecks.

pafurijaz, 2020-03-28:
I know very well how works the subdivision surfaces, and the Old FreeShip handheld this file very well. Only the newer Lazarus build is too slow. With original FreeShip version it took on loading only few second, and works very well.

markmal, 2020-03-28:
Good news. I've found where the issues were and fixed them. Now it works fast.
The fix is included into release 4.8.0.351. 
Check Releases page.

pafurijaz, 2020-03-29:
Thanks now is very fine on loading, thanks ;)

closed 2020-03-29 by markmal