markmal/freeship-plus-in-lazarus #59: IGES Symmetric export

pafurijaz, 2020-03-29:
First of all a great better exporting into IGES now I get all layers with color, very cool stuff, but still an old bug still present in all FreeShip version I have tried yours and non, In the layer proprieties there are the possibility select an option for visualization for non symmetric parts such as for example a single propeller, The exporting not take into consideration the option and export all into IGES joining the wrong generated surface, that happen even in lineseplan generation and you get an ugly result with non symmetric boat deck.

markmal, 2020-03-29:
I've started looking into ...
Is the problem - non-symmetric parts are double-mirrored in IGES?
 
I would appreciate if you send a simple example (not the huge one).
What CAD do use to load the IGES into?

pafurijaz, 2020-03-29:
Thanks, here is a file with a propeller and you can recognize that, when you export it into IGES you get a whole exported as mirrored objects far all layer even for non symmetric parts. Hi use various software and isn't a problem related to CADs but to the software exporting process.  I use Rhino gCAD3D a great software ta handle very well iges from FreeShip that was in past corrupted.

[Fishing_vessel_14mt_00.fbm.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/4399196/Fishing_vessel_14mt_00.fbm.zip)

pafurijaz, 2020-03-29:
An other bugs I thing now was impossible import VRML1 file format, here a file for testing,([C55-hull.wrl.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/4399379/C55-hull.wrl.zip))
 and if desire an application that menage very well VRML1 here a link https://www.patrickmin.com/meshconv/

markmal, 2020-03-29:
I've found gCAD3D 2.46 by Franz Reiter , not Rhino. Is it what do you use?
I just need something more or lees reliable to check results of export.

For VRML please open separate issue.

pafurijaz, 2020-03-30:
Sorry but Is the first i saw with many layer organized for better inspect the layers, anyway here below there are two simple file one cube an a propeller.
[Symmetric_Sample_file.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/4401326/Symmetric_Sample_file.zip)

markmal, 2020-03-30:
Thank you, that was helpful. I've fixed the issue. It will be included into new release soon.

pafurijaz, 2020-03-30:
> I've found gCAD3D 2.46 by Franz Reiter , not Rhino. Is it what do you use?
> I just need something more or lees reliable to check results of export.
gCAD3D is very good for that, I use Rhino. and if you need an other free software you can try CAD Assistant or FreeCAD.

closed 2020-03-31 by markmal