markmal/freeship-plus-in-lazarus #61: Main dimension precision

pafurijaz, 2020-04-06:
Hi in the latest version you have suppressed the zero in the input dialog box now is impossible to setup the right value of the boat because the field lacks in precision because the input have only the support for few floating, Already the old FreeShip is not able to support small objects, if you further reduce precision the accuracy will only be useful for making calculations on huge boats.

[image]
[image]
[image]

markmal, 2020-04-06:
For real boats precision of 0.1mm is not practical because anyway nobody can build a boat with such a precision. Unless you are making little models (e.g. 3-d printed). 
Also in many places in calculations tolerances are 1e-4. So except having precise outer dimensions you will not benefit from such precision in calculations results.

It is not a problem to make precision back to 4 decimals after the point. I'll do that. Next release.

pafurijaz, 2020-04-06:
Thanks, for response :), But anyway was impossible the calculation of hydrostatic for a small objects in fact is impossible take a plausible result from small boat, under 1 mt, Freeship merge the vertexes automatically when they are too close , and that make impossible to calculate such thing as for 3D small objects printed or other kind inside the boat.

markmal, 2020-04-06:
Please open another issues for:
- "impossible the calculation of hydrostatic for a small objects" and provide with example
- "Freeship merge the vertexes automatically when they are too close" - is it about export? Which export? I need an example too.

closed 2020-04-21 by pafurijaz