markmal/freeship-plus-in-lazarus #62: Do not show about dialog on each launch & do not ask "I agree"

Symbian9, 2020-04-06:
Please, disable showing "About" dialog on every FreeShip launch.

Also, remove this obsolete "I accept" checkbox as GNU GPL is a license and NOT an agreement

- https://twitter.com/RichFelker/status/1238113522187489283

> GPL is very clear that it's a license not an agreement, that you are under no obligation to agree to it if you don't intend to do anything that would be copyright infringement without the license, and that mere use is not something copyright can govern.

markmal, 2020-04-06:
Ok, I am not a lawyer, but there are following reasons for that:
1 - Someone can create a boat using FreeShip, that boat can sink and cause damages or even death. By accepting GPL (including items 15 and 16) a user acknowledges that I do not have any responsibilities for this.
2. GPL has terms and conditions. That means it is an agreement. It requires a user to accept them. In particular, my concern again is regarding sections 15 and 16. That is what the "I accept" is about.

15. Disclaimer of Warranty.
THERE IS NO WARRANTY FOR THE PROGRAM, TO THE EXTENT PERMITTED BY APPLICABLE LAW. EXCEPT WHEN OTHERWISE STATED IN WRITING THE COPYRIGHT HOLDERS AND/OR OTHER PARTIES PROVIDE THE PROGRAM “AS IS” WITHOUT WARRANTY OF ANY KIND, EITHER EXPRESSED OR IMPLIED, INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE. THE ENTIRE RISK AS TO THE QUALITY AND PERFORMANCE OF THE PROGRAM IS WITH YOU. SHOULD THE PROGRAM PROVE DEFECTIVE, YOU ASSUME THE COST OF ALL NECESSARY SERVICING, REPAIR OR CORRECTION.
16. Limitation of Liability.
IN NO EVENT UNLESS REQUIRED BY APPLICABLE LAW OR AGREED TO IN WRITING WILL ANY COPYRIGHT HOLDER, OR ANY OTHER PARTY WHO MODIFIES AND/OR CONVEYS THE PROGRAM AS PERMITTED ABOVE, BE LIABLE TO YOU FOR DAMAGES, INCLUDING ANY GENERAL, SPECIAL, INCIDENTAL OR CONSEQUENTIAL DAMAGES ARISING OUT OF THE USE OR INABILITY TO USE THE PROGRAM (INCLUDING BUT NOT LIMITED TO LOSS OF DATA OR DATA BEING RENDERED INACCURATE OR LOSSES SUSTAINED BY YOU OR THIRD PARTIES OR A FAILURE OF THE PROGRAM TO OPERATE WITH ANY OTHER PROGRAMS), EVEN IF SUCH HOLDER OR OTHER PARTY HAS BEEN ADVISED OF THE POSSIBILITY OF SUCH DAMAGES.

Symbian9, 2020-04-06:
> 2. GPL has terms and conditions.

There are NO "Terms and Conditions" in GNU GPL.

Also "Disclaimer of Warranty" and "Limitation of Liability" are NOT "Agreement", so users should NOT check/sign any "I accept".

markmal, 2020-04-06:
I need to read more on this or may be to consult with my lawyer.

If your concern is one click, you can turn off the splash with calling FreeShip with --nosplash* param.
Try FreeShip --help

markmal, 2020-04-06:
Yes, it is convoluted....

2. Basic Permissions.
... This License explicitly affirms your unlimited permission to run the unmodified Program. 

Means - only the License grants a user "unlimited permission to run the unmodified Program".
So to run it, he/she have to agree with the License.

But further I see:
9. Acceptance Not Required for Having Copies.
**You are not required to accept this License in order to receive or run a copy of the Program.** Ancillary propagation of a covered work occurring solely as a consequence of using peer-to-peer transmission to receive a copy likewise does not require acceptance.

and then again "15. Disclaimer of Warranty." and "16. Limitation of Liability."

If User can run Program without accepting the whole License and these items in particular, this means to me - the User does not want to acknowledge/accept my Limitation of Liability, means User makes **me** liable. It is wrong.

Symbian9, 2020-04-06:
> I'd prefer User accepts the License.

If user **downloaded* GNU GPL software its mean that user **already** (automatically) accepts the license; otherwise user should not download such software ­— this is what GNU GPL license about!

markmal, 2020-04-07:
1. It is nothing on GitHub that tells you that all you download is GNU GPL.
2. User can get the Program through other means, not nesesserily by download.
3. Program can be installed in multi-user or public computer. So users have no clue about how it was acquired and installed.

Symbian9, 2020-04-07:
> It is nothing on GitHub that tells you that all you download is GNU GPL.

Any downloads from GitHub are made manually by user (as there are no automatic downloads to user's PC) , so users always should check what they download.

Also each project on GitHub has license info in repo header (in this repo it represented as *"View license"*).

> User can get the Program through other means, not nesesserily by download.

Again, users should always check what they got from somewhere.

> Program can be installed in multi-user or public computer. So users have no clue about how it was acquired and installed.

By using any public computer users do it on own risks. Users should avoid use any public or any multi-user PC if they not sure what they do.

markmal, 2020-04-11:
Your reasoning does not sound convincing. I do not want to go to listing all permutations of how program can be installed and used. GPLv3 still has some ambiguities that can be resolved only in court. I don't want to be got into such situation.
If someone built a boat, then it sunk and someone died, or went disability, insurance lawyers will seek any possible opportunity to sue someone to get money from.
I do not want to be this "someone".
It is just one click altogether to accept license and to close the splash window.
If someone does not want to accept liabilities that come with designing a vessel, so he/she should not use this program.

closed 2020-04-13 by markmal