markmal/freeship-plus-in-lazarus #63: Always failed on opening "Open" dialog

Symbian9, 2020-04-06:
```
Program exception! 
Stacktrace:

Exception class: Exception
Message: Error reading from file_0 ! Load String
Position+Size:639902515 > 50288

  $00000000009D7F9E line 1873 of Units/FreeFileBuffer.pas
  $000000000096BC53 line 2249 of Units/FreeShip.inc
  $000000000092FB49 line 5126 of Units/FreeEdit.inc
  $0000000000538F15 line 1517 of Forms/Main.pas
  $00000000004C1874 line 680 of include/customform.inc
  $00000000004C1DF5 line 842 of include/customform.inc
  $0000000000472992 line 602 of ../inc/objpas.inc
  $00000000004AAA8D line 2243 of include/control.inc
  $000000000049D48C line 5413 of include/wincontrol.inc
  $00000000004C37EE line 1477 of include/customform.inc
  $00000000004A9196 line 1581 of include/control.inc
  $00000000004C776C line 3006 of include/customform.inc
  $000000000053925C line 1542 of Forms/Main.pas
  $0000000000539558 line 1577 of Forms/Main.pas
  $0000000000557FA0 line 124 of ../objpas/classes/action.inc
  $00000000005C20AD line 98 of include/containedaction.inc
  $00000000005C3FB5 line 246 of include/customaction.inc
```

markmal, 2020-04-06:
weird message "Error reading from file_0 ! Load String"
How do you start FreeShip, from shell, by clicking on program icon, by clicking on an ftm/fbm file icon?
Could you run it from shell without any parameters?

Symbian9, 2020-04-06:
> How do you start FreeShip, from shell, by clicking on program icon

Both ways.

> by clicking on an ftm/fbm file icon?

Not sure what you mean. "Open..." dialog failed before it show me file tree.

JFTR, If I try open file using associated program from file browser (by double click on file or using "Open file with...") — FreeShip open this file, BUT if then try to open "Open..." — dialog anyway failed.

markmal, 2020-04-06:
Now I see. Are there any other files (not FBM, FTM) in the directory?
It may happen that when Open dialog opens, it positions on a first file in the directory, it will try to show it in preview tile. For this it reads it, and fails.

I am not saying that it is wrong to have other files in the directory. Just need to reproduce your situation in my dev environment.

Symbian9, 2020-04-06:
> Are there any other files (not FBM, FTM) in the directory?

Lets be clear: I always use "clean install", so there NO any previously opened locations/files.

So, what folder it try to open on first launch "Open..." dialog? (I can't catch it because dialog failed before show me location)

> It may happen that when Open dialog opens, it positions on a first file in the directory, it will try to show it in preview tile. For this it reads it, and fails.

This sound like a bug: instead of just skip render preview full dialog crashed.

markmal, 2020-04-06:
Does "clean install" means that you also manually delete ~/Documents/FreeShip and ~/.config/FreeShip ?
apt does not do that because there are your projects and settings.
~/.config/FreeShip/FreeShip.ini

When Open dialog opens it reads LastDirectory from FreeShip.ini and tries to open it.
Chould you check what is there and what is in that fiolder?

I agree, I need to double check why exceptions in preview code are not captured and skipped. I remember I did that. I'll take a look.

Symbian9, 2020-04-06:
> Does "clean install" means that you also manually delete ~/Documents/FreeShip and ~/.config/FreeShip ?

Yes, of course.

markmal, 2020-04-07:
I tried to follow your trace dump but it does not show it happens in Open dialog. May be before.
I cannot reproduce this error. I removed ~/Documents/FreeShip and ~/.config/FreeShip.
No exception happened. ~/Documents/FreeShip is created automatically after program is started.

Please install new version 4.9.1.377 so stack dump line numbers match code I have now.
If exception happens please post the stack dump here.

markmal, 2020-04-08:
Hi Symbian9 
Did you have a chance to try with new version 4.9.1.377.
Any issues? Please let me know.

I am working on other issues and code is changing. I can make interpretation of your trace stack difficult.

Symbian9, 2020-04-08:
> Did you have a chance to try with new version 4.9.1.377.

Not yet, but would try it today.

Symbian9, 2020-04-09:
After 'clean install' upgrade to `4.9.1.377` issue seems solved.

So, I close this issue as for now.

closed 2020-04-09 by Symbian9