markmal/freeship-plus-in-lazarus #64: Bad whorse WRML import

pafurijaz, 2020-04-21:
With every version is getting worse VRML importing,
Here a mess file importing with FreeShip that i use some time to convert geometry to Nurbs. here what happened in this latest version..
[image]
and below the old FreeShip.
[image]
Below the file of the geometry.
[TestOBJ-wrl.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/4508970/TestOBJ-wrl.zip)

markmal, 2020-04-21:
I'll take a look.

markmal, 2020-04-22:
is car VRML both sided or one side?

pafurijaz, 2020-04-22:
The model i full without mirroring. because if you make half you have a creased edges in the middle and that avoid the pinch in the middle because is hard to get a tangent with other side.

markmal, 2020-04-22:
point tolerance was hardcoded to 1e-5. Some points that close enough were merged and this caused mess in faces.
I've made tolerance autodetected from precision of coords in VRML file.
Will be in next release.

pafurijaz, 2020-04-22:
wow the same problem due to same bug?

pafurijaz, 2020-06-01:
Now With latest version works very well,thanks :)

closed 2020-06-01 by pafurijaz