markmal/freeship-plus-in-lazarus #65: merging of very small objects

pafurijaz, 2020-04-21:
Here the example of that, below the files. a simple cube, in the original file there are 14 cubes but FreeShip recognize only 11 and merge the others.
[image]

[Cubes.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/4509127/Cubes.zip)
[Cube_test.ftm.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/4509133/Cube_test.ftm.zip)

markmal, 2020-04-22:
same story point tolerance was hardcoded to 1e-5. Some points that close enough were merged and this caused mess in faces.
I've made default tolerance 1e-6 and fixed sqrdistance comparison.
Now it is 14 cubes.
Will be in next release.

pafurijaz, 2020-04-22:
Great :)

pafurijaz, 2020-06-01:
Seem working with latest version ;) thanks!

closed 2020-06-01 by pafurijaz