markmal/freeship-plus-in-lazarus #66: Build fail with inkscape 1.0

mgian, 2020-05-09:
On Archlinux the build fail because a problem with inkscape, here is the log
gian@galactica ~/Devel/freeship-plus-in-lazarus (master)> make
for i in 16 24 32 48 256; do inkscape --export-png=freeship-$i.png -w $i -h $i FreeShip_icon.svg ; done
Unknown option --export-png=freeship-16.png
Unknown option --export-png=freeship-24.png
Unknown option --export-png=freeship-32.png
Unknown option --export-png=freeship-48.png
Unknown option --export-png=freeship-256.png
make: *** [Makefile:29: build-icons] Error 1
gian@galactica ~/Devel/freeship-plus-in-lazarus (master)> inkscape -v
Unknown option -v
gian@galactica ~/Devel/freeship-plus-in-lazarus (master)> inkscape -V
Inkscape 1.0 (4035a4fb49, 2020-05-01)
    Pango version: 1.44.7
gian@galactica ~/Devel/freeship-plus-in-lazarus (master)>

markmal, 2020-05-10:
Hi, I tried to make Makefile time ago. Then I abandoned it. I build using Lazarus IDE.

mgian, 2020-05-14:
Ok, fine. I will try 
Thanks

closed 2020-05-14 by mgian