markmal/freeship-plus-in-lazarus #67: Doesn't export line plan DXF files

pafurijaz, 2020-06-02:
Hi the new version don export the DXF file of linesplan

Below the output of Program exception! 
`Exception class: ERangeError
Message: Range check error
  $0000000000AA4547
  $0000000000AA32EC
  $0000000000AA06C1
  $000000000055BE00
  $00000000005C7D69
  $00000000005C9D9D
  $000000000055BA4C
  $00000000004ACDB7
  $00000000007489B9
  $0000000000746627
  $00000000004ABD04
  $00000000004ACA87
  $0000000000472B12
  $00000000004ABB0F
  $00000000004AA178
  $000000000049C743
  $000000000049DE31
  $0000000000000018
  $FFFFFFFFFFFFFFE8
  $0000000000FC96C0
  $000000000100B540
  $0000000000000000
  $0000000000000000
  $0000000000000000
  $0000000001014840`

And this is the console output.
`~$ FreeShip
FreeShip in Lazarus
Compiled at 2020/06/01 08:35:21
Compiler version: 3.0.4
Target CPU: x86_64
Target OS: Linux
FreeShip Product version: 5.0.30
FreeShip Program version: 5.0.30.422
Last Git Change Revision: 407
Last Git Commit: 24498f2345f9878dfbd43406169a4753930575ab

(FreeShip:3553): GLib-GIO-CRITICAL **: 10:35:08.397: g_dbus_proxy_new: assertion 'G_IS_DBUS_CONNECTION (connection)' failed

(FreeShip:3553): GLib-GIO-CRITICAL **: 10:35:08.397: g_dbus_proxy_new: assertion 'G_IS_DBUS_CONNECTION (connection)' failed

(FreeShip:3553): GLib-GIO-CRITICAL **: 10:35:08.397: g_dbus_proxy_new: assertion 'G_IS_DBUS_CONNECTION (connection)' failed
Exception class: EAccessViolation Message: Access violation
Program exception! 
Stacktrace:

Exception class: ERangeError
Message: Range check error
  $0000000000AA4547
  $0000000000AA32EC
  $0000000000AA06C1
  $000000000055BE00
  $00000000005C7D69
  $00000000005C9D9D
  $000000000055BA4C
  $00000000004ACDB7
  $00000000007489B9
  $0000000000746627
  $00000000004ABD04
  $00000000004ACA87
  $0000000000472B12
  $00000000004ABB0F
  $00000000004AA178
  $000000000049C743
  $000000000049DE31
  $0000000000000018
  $FFFFFFFFFFFFFFE8
  $0000000000FC96C0
  $000000000100B540
  $0000000000000000
  $0000000000000000
  $0000000000000000
  $0000000001014840
  $0000000001014818
  $0000000000000000
  $000000000147F260
  $0000000000000000
  $00000000004723B0
  $00000000004725A0
  $0000000000472620
  $0000000000472910
WARNING: TToolBar.Destroy with LCLRefCount>0. Hint: Maybe the component is processing an event?
[TGtk2WidgetSet.Destroy] WARNING: There are 32 unreleased GDIObjects, a detailed dump follows:
[TGtk2WidgetSet.Destroy]   GDIOs: 0000000002D7AE08 0000000002E24A78 0000000002E892A8 0000000002E55A38 0000000002E54708 0000000002E558A8 0000000002E54898
[TGtk2WidgetSet.Destroy]   gdiBitmap: 32
`
The file used for the export
[Knud_Reimers_Powerboat_72ft_FBL.fbm.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/4715905/Knud_Reimers_Powerboat_72ft_FBL.fbm.zip)

froussel, 2020-06-07:
Me also 
With kabeja.exe  

Parsing:C:\Users\admin\Documents\FreeShip\Export\buttocks.dxf
org.kabeja.parser.ParseException: Line: 18 unsupported groupcode: false for value:0,0
	at org.kabeja.parser.DXFParser.parseGroup(DXFParser.java:205)
	at org.kabeja.parser.DXFParser.parse(DXFParser.java:141)
	at org.kabeja.ui.impl.ProcessingRunViewComponent.parseFile(ProcessingRunViewComponent.java:340)
	at org.kabeja.ui.impl.ProcessingRunViewComponent.processInput(ProcessingRunViewComponent.java:406)
	at org.kabeja.ui.impl.ProcessingRunViewComponent.chooseInput(ProcessingRunViewComponent.java:328)
	at org.kabeja.ui.impl.ProcessingRunViewComponent$1$1.run(ProcessingRunViewComponent.java:179)
	at java.lang.Thread.run(Unknown Source)
Caused by: java.lang.NumberFormatException: For input string: "0,0"
	at sun.misc.FloatingDecimal.readJavaFormatString(Unknown Source)
	at sun.misc.FloatingDecimal.parseDouble(Unknown Source)
	at java.lang.Double.parseDouble(Unknown Source)
	at org.kabeja.parser.DXFValue.getDoubleValue(DXFValue.java:51)
	at org.kabeja.parser.entities.DXFPolylineHandler.parseGroup(DXFPolylineHandler.java:128)
	at org.kabeja.parser.DXFEntitiesSectionHandler.parseGroup(DXFEntitiesSectionHandler.java:80)
	at org.kabeja.parser.DXFParser.parseGroup(DXFParser.java:199)
	... 6 more

markmal, 2020-06-07:
Sorry I don't support kabela.

> Caused by: java.lang.NumberFormatException: For input string: "0,0"
it looks like java is also confused with comma as a dec separator.

markmal, 2020-06-09:
I can open your file Knud_Reimers_Powerboat_72ft_FBL.fbm.zip with version 5.0.33.431

pafurijaz, 2020-06-09:
Why you can't open this file is saved with your build? and the workflow is simple I go in the project menu and then in the linesplan and save the file from this.
[image]
[image]
Saved now with latest build you provided..
[Knud_Reimers_Powerboat_72ft_FBL.fbm.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/4753363/Knud_Reimers_Powerboat_72ft_FBL.fbm.zip)
and here something new seem your freeShip Build save an older file than mine the original FreeShip latest shared on network 3.43 but I don't have any source code, maybe I can share my build with you if desire.
[image]
[image]
FreeShip_file
[Knud_Reimers_Powerboat_72ft_FBL_old_FreeShip.fbm.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/4753414/Knud_Reimers_Powerboat_72ft_FBL_old_FreeShip.fbm.zip)

pafurijaz, 2020-06-10:
Now with this latest build, export again the DXF linesplan thanks :)

closed 2020-06-10 by pafurijaz