markmal/freeship-plus-in-lazarus #68: Shade Under water

pafurijaz, 2020-06-02:
The hull shade under water not show the color selected but a bit darker main hull color, work only with windows lines plan.
Below some images
[image]
[image]

markmal, 2020-06-02:
Underwater coloring is changed to translucent mode to be more natural. So underwater color now is blend of layer color and underwarer color.

pafurijaz, 2020-06-02:
OK thanks, for response, but very low contrast and is hard to see but anyway is good knowing was not a bug. :)

closed 2020-06-02 by pafurijaz

markmal, 2020-06-02:
I'll check if I can make something to improve contrast.

pafurijaz, 2020-06-02:
very nice, even for who has visual problem..