markmal/freeship-plus-in-lazarus #70: open file ftm impossible

froussel, 2020-06-06:
[image]

froussel, 2020-06-06:
FTM created with 5.0.30.422 open possible but file, open make
[image]

ftm created with 4.9.2.380 can't open with 5.0.30 and not with 4.9.2

height hours lost  :-(((((

froussel, 2020-06-07:
It was in the preferences.ini folder that there was something that made opening a file trigger the software closing.

markmal, 2020-06-07:
Hi froussel
it looks like your numeric settings have comma as a decimal separator. Please confirm.
What is your OS?

markmal, 2020-06-09:
anyway, I've made program to accept any decimal separator, dot or comma.

froussel, 2020-06-11:
Hello,
I loaded version 422 and opened with the file
[http://dl.free.fr/v1E9k7Oyb]
dxf export does not work properly

Besides, I would like to be able to export the shell to FreeCAD
I get there by exporting a stl but then in freeCAD it's a pain because we only have mesh.
it would be interesting if the data contained in the 2D dxf is presented in the form of files
[https://tracfoil.com/airfoils/index.php?page=fr_file] .DAT or .COR
Because then there are freecad users of this type of files who have made macros to transform into B-splines
Or for integrate into freeCAD an export of its kind
[https://mon-partage.fr/f/BURzxfbo/]

markmal, 2020-06-11:
I've loaded 6b.fbm, exported into 3D DXF mesh and loaded the 6b.dxf into https://viewer.autodesk.com
No failures. The model looks correct. 
Please explain why dxf export does not work properly? What is wrong?

I'll take a look into export to .DAT.

Do you still have the initial problem "open file ftm impossible" ?

froussel, 2020-06-11:
Hello,

See files

Regards

fr

0
SECTION
2
ENTITIES
0
POLYLINE
8
Buttocks
62
135
66
1
0
VERTEX
8
Buttocks
10
0.0
20
0.0
0
VERTEX
8
Buttocks
10
0.0
20
0.0
0
VERTEX
8
Buttocks
10
0.0
20
0.0
0
VERTEX
8
Buttocks
10
0.0
20
0.0
0
VERTEX
8
Buttocks
10
0.0
20
0.0
0
VERTEX
8
Buttocks
10
0.0
20
0.0
0
VERTEX
[... 11593 lines cut ...]
Buttocks
10
114.26653
20
14.25105
0
VERTEX
8
Buttocks
10
114.26653
20
14.25105
0
VERTEX
8
Buttocks
10
114.26653
20
14.25105
0
VERTEX
8
Buttocks

markmal, 2020-06-11:
The text of DXF is truncated.
Please attach the zip here.
How did you export into the file? Which menu item?

froussel, 2020-06-12:
Hello,

In zip

Regards

fr

markmal, 2020-06-12:
I do not see any zip. This is how zip should be attached.
[6b.dxf.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/4770651/6b.dxf.zip)

I will close the case if you will not cooperate. I assume the initial problem is fixed.

froussel, 2020-06-12:
[bug.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/4770908/bug.zip)

markmal, 2020-06-12:
looking ...

markmal, 2020-06-12:
please validate output and let me know if it is correct.
[6b-dxf-2d-poly.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/4773689/6b-dxf-2d-poly.zip)

froussel, 2020-06-13:
an export to .dat: wonderful
for dxf it's almost perfect
buttovks1: two layers welded
stations1: one segment too many
waterlines1 the back is weird and an extra layer but in 4.3 too
the whole although perfectible is perfectly usable to be corrected
[bug1.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/4774133/bug1.zip)

markmal, 2020-06-13:
Waterlines weird is because there is deck. Those weird WLs are on deck. In FreeShip turn all visibility off except for waterlines, look in 3-D view.
Please validate this
[6b-dxf-2d-poly.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/4775365/6b-dxf-2d-poly.zip)

froussel, 2020-06-13:
[buttock2.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/4775428/buttock2.zip)
OK For your comments on the waterlines
stations: perfect
buttocks: the two curves are not separated

markmal, 2020-06-14:
What do you mean by "the two curves are not separated"? There are 6 separate polylines in buttocks.dxf

froussel, 2020-06-14:
Each of the erased curves can be edited, they come as independent
The 2 which are in the previous zip are welded by the bridge line.
When will you publish a version that fixes as you do now that I can test on other fbm files as well?

markmal, 2020-06-14:
I cannot reproduce your bug. 
I've opened buttocks.dxf in LibreCAD 2.1.2, 
Selected each buttock - each curve was selected separately.
Selected and deleted each buttock - each curve was deleted separately.
What CAD do you use? May be issue is there?

froussel, 2020-06-16:
Yes it's perfect in librecad
I use http://www.iprocam.com/download/setup-ninos-v5.exe
which without license allows to work on the dxf and which I use with license for my CNC
The buttocks2 can be used by exploding and then grouping the points together so the problem is minor for me.
The ideal would be to present the dxf in different layers as in the file below, it should not be very difficult to integrate
[buttocks3.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/4784392/buttocks3.zip)
Each curve is in a different layer and therefore easier to isolate

froussel, 2020-06-25:
I just tested v 5.0.35.462
The corrections have not been incorporated?
[dxf.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/4829578/dxf.zip)

markmal, 2020-06-26:
DDX output as polylines was corrected. 
Buttocks, stations and WLs are output as DDX polylines. It works for LibreCAD.
The issue is with Ninos that breaks polylines.
I'll add layers later.

froussel, 2020-06-27:
DDX ???    
I'm talking about 2D export and I can tell you that version 438 has not been corrected

[dxf1r.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/4840044/dxf1r.zip)

markmal, 2020-06-27:
Sorry DXF. 
You should use v 5.0.35.462. I do not see issues there.
[dxf-2d.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/4840547/dxf-2d.zip)

Please check DXFs with layers. 
[dxf-2d-with-layers.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/4840548/dxf-2d-with-layers.zip)

froussel, 2020-06-27:
With layers :
The result is very satisfactory. But for a nitpicker like me you can still improve
[little thing.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/4840834/little.thing.zip)
 
without layers 
 the two curves in buttocks are grouped as in the other case.
The rest is error free
If possible I prefer with layers 
Beau travail !

froussel, 2020-06-27:
> You should use v 5.0.35.462. I do not see issues there.

I know i'm wrong but i'm using windows and pack doesn't exist

markmal, 2020-06-30:
I'll check dxf with layers. Please lets us both use LibreCAD as a common ground.

For Windows version please download FreeShip-5.0.35.461_x64.msi. from https://github.com/markmal/freeship-plus-in-lazarus/releases . Open Assets there
.
When you start 438 it should show that 5.0.35.461 update is available. Alternatively open menu Help -> Check Updates.

froussel, 2020-06-30:
With 432 I have :
Exception class: EInOutError Message: Could not initialize OpenSSL library
Possible libssl v1.1 compatibility issue.
libssl v1.1 is found.
FreeShip is currently compatible with libssl v1.0 only.
You have to check for updates traditional way at Download page. ...

With 461 :
Exception class: EInOutError Message: Could not initialize OpenSSL library
Possible libssl v1.1 compatibility issue.
libssl v1.1 is found.
FreeShip is currently compatible with libssl v1.0 only.
You have to check for updates traditional way at Download page. ...

And in console : 
Exception class: EInOutError Message: Could not initialize OpenSSL library
[WARNING] GetPosition called without handle for PairSplitterFilesAndPreview(TPairSplitter)

No layers :-( 
dxf-2d-with-layers.zip it was more cool, even in the current state ! 
 "lets us both use LibreCAD as a common ground" : no problem.

markmal, 2020-06-30:
What is your OS?

Also I've found error when Check updates cannot find newer version. Fixing.

froussel, 2020-06-30:
win 10

markmal, 2020-07-07:
Regarding issue with layer 0 in LibreCAD, it is most likely a bug in LibreCAD.
I've checked DXF files in https://viewer.autodesk.com. It does not show any layer "0".
Only good layers.

markmal, 2020-07-13:
please try new version

closed 2024-04-18 by markmal