markmal/freeship-plus-in-lazarus #72: Design hydrostatic calc hull Errors

pafurijaz, 2020-06-10:
Hi I get an errors on this hull FreeShip say is to small or to large, but the same works well with older FreeShip the standard releases, not your build, the older application, can you look on that?
[Hull_TC_67.ftm.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/4757910/Hull_TC_67.ftm.zip)

markmal, 2020-06-12:
Wrong data in the file. All faces are inverted. They need to be inverted back, so normals should be pointing outside.
Use menu Select -> Select All.
menu Face -> Invert
After this Design hydrostatics will be calculated correctly.
Save model after this.

pafurijaz, 2020-06-12:
Yes Is true I checked and many faces are flipped inside, now the hydrostatic is performed  properly, this is something to remember. and something to add in the Manual.

closed 2020-06-12 by pafurijaz