markmal/freeship-plus-in-lazarus #73: Error and crash on closing Hydrostatic panel

pafurijaz, 2020-06-12:
Below the console the full console output and hull and a short video

```
FreeShip
FreeShip in Lazarus
Compiled at 2020/06/09 20:50:17
Compiler version: 3.0.4
Target CPU: x86_64
Target OS: Linux
FreeShip Product version: 5.0.35
FreeShip Program version: 5.0.35.438
Last Git Change Revision: 422
Last Git Commit: bb63507e5b02d259b557593d4e3cf62df6343309

(FreeShip:20913): GLib-GIO-CRITICAL **: 16:35:35.327: g_dbus_proxy_new: assertion 'G_IS_DBUS_CONNECTION (connection)' failed

(FreeShip:20913): GLib-GIO-CRITICAL **: 16:35:35.345: g_dbus_proxy_new: assertion 'G_IS_DBUS_CONNECTION (connection)' failed

(FreeShip:20913): GLib-GIO-CRITICAL **: 16:35:35.345: g_dbus_proxy_new: assertion 'G_IS_DBUS_CONNECTION (connection)' failed
Program exception! 
Stacktrace:

Exception class: EGridException
Message: Index Out of range Cell[Col=0 Row=7]
  $00000000007CA2D7
  $00000000007CDDDD
  $0000000000A9323C
  $00000000004AE832
  $00000000006C5899
  $00000000006C7C34
  $0000000000472BF2
  $00000000004AD5AF
  $00000000004ABC18
  $000000000049E1E3
  $000000000049F8D1
  $00000000007976E2
  $0000000000777DB5
  $00000000007877FF
  $00000000007879CB
  $00007FBD118DC38B
  $00007FFF5171D7E0
  $0000000000538ECA
  $00007FFF5171D7E0
  $00000000018225D8
  $0000000001FFAEB8
  $0000000000000000
  $00007FFF5171D820
  $00000000004D35E2
  $00000000007878E0
  $00007FFF5171DC20
  $00007FFF5171DC80
  $000000005171D800
  $00000000016CAB78
  $0000000000000000
  $00007FFF5171D980
WARNING: TPanel.Destroy with LCLRefCount>0. Hint: Maybe the component is processing an event?
WARNING: TMenuItem.Destroy with LCLRefCount>0. Hint: Maybe the component is processing an event?
[TGtk2WidgetSet.Destroy] WARNING: There are 32 unreleased GDIObjects, a detailed dump follows:
[TGtk2WidgetSet.Destroy]   GDIOs: 0000000001DEEF18 0000000001CE11C8 0000000001E38098 0000000001DE81B8 0000000001DE6E88 0000000001E03D78 0000000001BE8CA8
[TGtk2WidgetSet.Destroy]   gdiBitmap: 32

```
Zip file with a video..
[Hull_Yacht_67mt.fbm.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/4771479/Hull_Yacht_67mt.fbm.zip)

pafurijaz, 2020-06-12:
But I  put an other concern here, and I don't know is this is due to the Lazarus code, but the cross curves for this hull took a very long time before was performed.

markmal, 2020-06-12:
please execute 
**locale -k LC_NUMERIC**

pafurijaz, 2020-06-13:
Here the output for
```

$ locale -k LC_NUMERIC
decimal_point=","
thousands_sep="."
grouping=3;3
numeric-decimal-point-wc=44
numeric-thousands-sep-wc=46
numeric-codeset="UTF-8"

```

markmal, 2020-06-15:
fixed

closed 2020-06-15 by pafurijaz