markmal/freeship-plus-in-lazarus #76: Resistence error for planinig hull

pafurijaz, 2020-06-16:
Hi this is a file with resistance calculated in the old FreeShip, but when I try to recalculate inside your Build this crash immediately.
File below please test with other method for resistance perform. Thanks ;)
```
Program exception! 
Stacktrace:

Exception class: EInOutError
Message: Access denied
  $0000000000B86A17 line 2253 of Forms/FreeResistance_PlaningDlg.pas
  $0000000000B8C782 line 2678 of Forms/FreeResistance_PlaningDlg.pas
  $00000000004AE832 line 2911 of include/control.inc
  $00000000004F7DDA line 55 of include/buttoncontrol.inc
  $00000000004F67A1 line 257 of include/customcheckbox.inc
  $00000000004F63A7 line 162 of include/customcheckbox.inc
  $0000000000B8B41C line 2516 of Forms/FreeResistance_PlaningDlg.pas
  $000000000096541D line 9041 of Units/FreeEdit.inc
  $0000000000543BF7 line 2912 of Forms/Main.pas
  $0000000000564480 line 124 of ../objpas/classes/action.inc
  $00000000005D03E9 line 98 of include/containedaction.inc
  $00000000005D241D line 246 of include/customaction.inc
  $00000000005640CC line 49 of ../objpas/classes/action.inc
  $00000000005CB019 line 86 of include/menuitem.inc
  $00000000005CB94A line 293 of include/menuitem.inc
  $0000000000472BF2 line 602 of ../inc/objpas.inc
  $00000000007976F9 line 114 of lclmessageglue.pas
  $00007FFD8FBBF370
  $0000000000538ECA line 794 of Forms/Main.pas
  $00007FFD8FBBF370
  $0000000001B4C498
  $00000000022A7BE8
  $0000000000000000
  $00007FFD8FBBF3B0
  $00000000004D35E2 line 1258 of include/application.inc
  $0000000000000000
  $0000000000000001
  $00007FFD8FBBF760
  $000000008FBBF400
  $00000000019F2B78
  $0000000000000000
  $00007FFD8FBBF510
  $000000000079775C line 116 of lclmessageglue.pas
```
[Concept_hull_00.fbm.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/4789320/Concept_hull_00.fbm.zip)

markmal, 2020-06-17:
looking ...

markmal, 2020-07-13:
please try new version

froussel, 2020-07-14:
I'll see if it happens again

pafurijaz, 2020-07-21:
bug again with latest build, 
```
Program exception! 
Stacktrace:

Exception class: EInOutError
Message: File not found
  $0000000000B8B5EB line 2282 of Forms/FreeResistance_PlaningDlg.pas
  $0000000000B74B64 line 436 of Forms/FreeResistance_PlaningDlg.pas
  $0000000000B914A2 line 2716 of Forms/FreeResistance_PlaningDlg.pas
  $00000000004AE8E2 line 2911 of include/control.inc
  $00000000004F7E8A line 55 of include/buttoncontrol.inc
  $00000000004F6851 line 257 of include/customcheckbox.inc
  $00000000004F6457 line 162 of include/customcheckbox.inc
  $0000000000B9013C line 2553 of Forms/FreeResistance_PlaningDlg.pas
  $000000000096837D line 9152 of Units/FreeEdit.inc
  $0000000000543DD7 line 2910 of Forms/Main.pas
  $00000000005647F0 line 124 of ../objpas/classes/action.inc
  $00000000005D0759 line 98 of include/containedaction.inc
  $00000000005D278D line 246 of include/customaction.inc
  $000000000056443C line 49 of ../objpas/classes/action.inc
  $00000000005CB389 line 86 of include/menuitem.inc
  $00000000005CBCBA line 293 of include/menuitem.inc
  $0000000000472CA2 line 602 of ../inc/objpas.inc
  $0000000000B74B64 line 436 of Forms/FreeResistance_PlaningDlg.pas
  $0000000000B914A2 line 2716 of Forms/FreeResistance_PlaningDlg.pas
  $00000000004AE8E2 line 2911 of include/control.inc

```

pafurijaz, 2020-07-21:
But at the end after errors it's works without crashing. 
[image]

closed 2020-09-21 by pafurijaz