markmal/freeship-plus-in-lazarus #77: littles things and big question

froussel, 2020-06-29:
For the little things:
When we enlarge a window often it stays behind (image 1)
When you change the language, some menus remain, you have to restart so that everything is up to date.
[plant.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/4847345/plant.zip)
When you move the shaded view, from time to time the software closes, so backup very closely. We could automatically save a bak file recorded after each operation which would be deleted when saving.

The big question: would it be possible to be able to observe the water lines when the boat is heeling?
For now, I reflect the hull (the weight of the displacement is doubled, no problem) then I transform with a rotation: fbm file
the draft remains that of the start so the waterline becomes that of a different weight, you have to grope but you can only see the water lines by multiplying the draft.
Would it be a big job to have in the shaded view a mouse action + key pressed which would show the water lines with equal displacement ... First for the lodging then with a rotation with the transverse axis .. .

I know it's not Christmas, but I still believe in Saint Nicolas :-)
[plant.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/4847329/plant.zip)

markmal, 2020-06-30:
1. Workaround - the internal window needs to be selected before you click enlarge button. I'll check if I can fix this.
2. I am in a process to change internalization/localization to standard approach. Currently there are 3 approaches mixed, so all kinds of issues with translations possible. Yes you better restart after you change language.
3. When you move shaded view, does it close with an error? I better fix this issue. But I need an error stack.
4. I will take a look.

Could you please next time open issues as separate ones? It will better for tracking.

froussel, 2020-06-30:
3 : no error

markmal, 2020-06-30:
3: I've found the error. Fixed.

markmal, 2020-07-13:
please try new version.

regarding "The big question: would it be possible to be able to observe the water lines when the boat is heeling?" -  needs more research

froussel, 2020-07-14:
Yes I also think that it requires a lot of research ...
To do when you want to give yourself a challenge

Should I close the issue?

alexandercardozo98, 2022-04-18:
Consulta, hay algún otro formato de .txt el cual puedas cargar los puntos en el programa, a mi por ejemplo me lo carga pero a la hora de editarlo me sale un mensaje extenso y se cierra el programa, me ayudarían por favor, desde ya muchas gracias y disculpen las molestias!

alexandercardozo98, 2022-04-19:
el mensaje de error que me sale es el siguiente luego de haber cargado la superficie con un archivo .txt

Program exception! 
Stacktrace:

Exception class: EInvalidOp
Message: Invalid floating point operation
  $0000000100285FB0 line 654 of Units/FreeSpline.inc
  $0000000100309A77 line 78 of Units/FreeMarker.inc
  $0000000100377057 line 3376 of Units/FreeShip.inc
  $0000000100377F17 line 3548 of Units/FreeShip.inc
  $0000000100382EF4 line 823 of Forms/freehullformwindow_panel.pas
  $000000010027071E line 1714 of Units/FreeViewport.inc
  $000000010003A283 line 2281 of include/control.inc
  $000000010003A949 line 2460 of include/control.inc
  $00000001000100F5
  $000000010003A173 line 2241 of include/control.inc
  $000000010002D89D line 5411 of include/wincontrol.inc
  $0000000100206C3D line 112 of lclmessageglue.pas
  $000000010019EFB0 line 2515 of win32callback.inc
  $000000010019F7DC line 2677 of win32callback.inc
  $00007FF8B113E858
  $00007FF8B113E299
  $00000001001A09F4 line 407 of win32object.inc
  $0000000100309A77 line 78 of Units/FreeMarker.inc
  $0000000100377057 line 3376 of Units/FreeShip.inc
  $0000000100377F17 line 3548 of Units/FreeShip.inc
  $0000000100382EF4 line 823 of Forms/freehullformwindow_panel.pas
  $000000010027071E line 1714 of Units/FreeViewport.inc
  $000000010003A283 line 2281 of include/control.inc
  $000000010003A949 line 2460 of include/control.inc
  $00000001000100F5
  $000000010003A173 line 2241 of include/control.inc
  $000000010002D89D line 5411 of include/wincontrol.inc
  $0000000100206C3D line 112 of lclmessageglue.pas
  $000000010019EFB0 line 2515 of win32callback.inc
  $000000010019F7DC line 2677 of win32callback.inc
  $00007FF8B113E858
  $00007FF8B113E299
  $00000001001A09F4 line 407 of win32object.inc

closed 2024-04-18 by markmal