markmal/freeship-plus-in-lazarus #85: Crash during Linesplan DXF export

CzifraP, 2021-01-31:
Hello,

when I tried to export the Linesplan of my hull to dxf I got this message below.

Program exception! 
Stacktrace:

Exception class: ERangeError
Message: Range check error
  $0000000000A8D841  ADDSPLINE,  line 1834 of Forms/FreeLinesplanFrme.pas
  $0000000000A8BD21  ADDINTERSECTION,  line 1939 of Forms/FreeLinesplanFrme.pas
  $0000000000A8A5F9  EXPORTDXFEXECUTE,  line 2075 of Forms/FreeLinesplanFrme.pas
  $000000000055EB9C
  $00000000005CE4E9  EXECUTE,  line 246 of include/customaction.inc
  $000000000055E892
  $0000000000729359  CLICK,  line 919 of include/toolbutton.inc
  $0000000000726F7F  MOUSEUP,  line 175 of include/toolbutton.inc
  $00000000004A69D4  DOMOUSEUP,  line 2301 of include/control.inc
  $00000000004A7755  WMLBUTTONUP,  line 2783 of include/control.inc
  $000000000046EE3A
  $00000000004A4E08  PERFORM,  line 1581 of include/control.inc
  $0000000000496FF4  ISCONTROLMOUSEMSG,  line 4755 of include/wincontrol.inc
  $000000000049877F  WNDPROC,  line 5376 of include/wincontrol.inc
  $0000000000776738  DELIVERMESSAGE,  line 112 of lclmessageglue.pas
  $00000000007574F5  DELIVERMESSAGE,  line 3780 of gtk2proc.inc
  $0000000000766DBC  DELIVERMOUSEUPMESSAGE,  line 2209 of gtk2callback.inc
  $0000000000A8BD21  ADDINTERSECTION,  line 1939 of Forms/FreeLinesplanFrme.pas
  $0000000000A8A5F9  EXPORTDXFEXECUTE,  line 2075 of Forms/FreeLinesplanFrme.pas
  $000000000055EB9C
  $00000000005CE4E9  EXECUTE,  line 246 of include/customaction.inc
  $000000000055E892
  $0000000000729359  CLICK,  line 919 of include/toolbutton.inc
  $0000000000726F7F  MOUSEUP,  line 175 of include/toolbutton.inc
  $00000000004A69D4  DOMOUSEUP,  line 2301 of include/control.inc
  $00000000004A7755  WMLBUTTONUP,  line 2783 of include/control.inc
  $000000000046EE3A
  $00000000004A4E08  PERFORM,  line 1581 of include/control.inc
  $0000000000496FF4  ISCONTROLMOUSEMSG,  line 4755 of include/wincontrol.inc
  $000000000049877F  WNDPROC,  line 5376 of include/wincontrol.inc
  $0000000000776738  DELIVERMESSAGE,  line 112 of lclmessageglue.pas
  $00000000007574F5  DELIVERMESSAGE,  line 3780 of gtk2proc.inc
  $0000000000766DBC  DELIVERMOUSEUPMESSAGE,  line 2209 of gtk2callback.inc

CzifraP, 2021-01-31:
I found the cause of the exception here.

ADDSPLINE, line 1824 of Forms/FreeLinesplanFrme.pas NParams variable was used in the for loop instead of the right NValues variable.

markmal, 2023-01-12:
Thanks for finding. I'll check.

CzifraP, 2023-01-14:
Hello FiftySixFactory

I didn't fixed on the git.
I downloaded the source code from GIT. I installed the Lasarus IDE and I ran the code from the ide in debugging mode. I found the issue.  I fixed in my source and I ran the Freeship from the Lasarus IDE. You can do the sam until markmal fix the code and build the packages again. 
P.s. I didn't do any git trick because i'm not realy familiar with it. I just wrote this message.

closed 2024-04-18 by markmal