markmal/freeship-plus-in-lazarus #91: IGES export bug in Windows (version 5.0.30.402)

albalfer, 2021-12-09:
Hi @markmal and @speters !

Awesome software for creating parametric surfaces from meshes.

I am experiencing a random error during export to IGES. It seems that at some parts of the file it writes "," instead of "." for decimal values thus creating incorrect vectors. The error happens randomly just on some files. It reproduces with executions of the same file on the same computer. I managed to sort it out in most of the cases either running the software on a different computer or scaling the object. I attach two examples:
- one example where I mananged to sort it out (running it on different computer), where you can see the bug (forefoot_L_lat.zip) one file is correct and the other incorrect.
- another example that I have not managed to sort out after trying to run it in 3 different computers (midfoot_L_lat.zip).

Both files where originally imported as wrl meshes.

Thanks in advance!

BR,

files attached:
[midfoot_L_lat.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/7684214/midfoot_L_lat.zip)
[forefoot_L_lat.zip](https://github.com/markmal/freeship-plus-in-lazarus/files/7684215/forefoot_L_lat.zip)

albalfer, 2021-12-14:
Hi again,
I managed to sort out the problem consistently by changing my regional configuration from Spanish (comma for deciamals and dot for thousands) to the opposite. Now it works 100% of the time. With the Spanish regional configuration it failed around 1 out of 3 files.
I hope this information helps to resolve the issue.
BR,

markmal, 2021-12-27:
Hi albafer, thanks for the information. Yes, decimal digits sometimes can be written according to regional configuration.
I was, and currently am, busy. Probably will take a look there later.
Good to know you have found the workaround.

closed 2024-04-23 by markmal